wasm-micro-runtime/wasm-micro-runtime #3858: WAMR fails names.wast spec test

sjamesr, 2024-10-15:
### Subject of the issue

WAMR currently fails the names.wast spec test, specifically because the spec permits module exports with names containing NUL bytes, but WAMR does not.

### Test case

Run `names.wast` through iwasm using the spec test runner scripts.

### Your environment

* Linux, x86_64
* WAMR head, fast interpreter

### Steps to reproduce

* Run the `names.wast` spec through the spec test runner. Specifically, we have:

```
  ;; Test the C0 control codes.
  (func (export "\00\01\02\03\04\05\06\07\08\09\0a\0b\0c\0d\0e\0f") (result i32) (i32.const 22))
```

and later

```
(assert_return (invoke "\00\01\02\03\04\05\06\07\08\09\0a\0b\0c\0d\0e\0f") (i32.const 22))
```

### Expected behavior

The tests should pass.

### Actual behavior

This fails, because the `wasm_lookup_function` function stops searching when it sees the null byte at the beginning of the function.

### Extra Info

I have the beginnings of a proposed fix which adds a `size_t name_len` field wherever export names are stored. This will change the AOT module representation, I don't know how much of a big deal this is.

I am curious if there is any interest in pursuing this change, or if WAMR deliberately does not want to support this aspect of the specification.

```
diff --git a/core/iwasm/aot/aot_runtime.c b/core/iwasm/aot/aot_runtime.c
index 63a3c83c..71e3501d 100644
--- a/core/iwasm/aot/aot_runtime.c
+++ b/core/iwasm/aot/aot_runtime.c
@@ -2185,6 +2185,22 @@ aot_lookup_function(const AOTModuleInstance *module_inst, const char *name)
     return NULL;
 }
 
+AOTFunctionInstance *
+aot_lookup_function_raw(const AOTModuleInstance *module_inst,
+                        const char *name, size_t name_len)
+{
+    uint32 i;
+    AOTFunctionInstance *export_funcs =
+        (AOTFunctionInstance *)module_inst->export_functions;
+
+    for (i = 0; i < module_inst->export_func_count; i++)
+        if (export_funcs[i].func_name_len == name_len
+            && !memcmp(export_funcs[i].func_name, name, name_len))
+            return &export_funcs[i];
+    return NULL;
+}
+
+
 #ifdef OS_ENABLE_HW_BOUND_CHECK
 static bool
 invoke_native_with_hw_bound_check(WASMExecEnv *exec_env, void *func_ptr,
[... 206 lines cut ...]
 WASMFunctionInstance *
 wasm_lookup_function(const WASMModuleInstance *module_inst, const char *name);
 
+WASMFunctionInstance *
+wasm_lookup_function_raw(const WASMModuleInstance *module_inst,
+                         const char *name, size_t name_len);
+
 WASMMemoryInstance *
 wasm_lookup_memory(const WASMModuleInstance *module_inst, const char *name);
 
diff --git a/wamr-compiler/CMakeLists.txt b/wamr-compiler/CMakeLists.txt
index 2ab0462b..210ff481 100644
--- a/wamr-compiler/CMakeLists.txt
+++ b/wamr-compiler/CMakeLists.txt
@@ -42,7 +42,7 @@ add_definitions(-DWASM_ENABLE_SIMD=1)
 add_definitions(-DWASM_ENABLE_REF_TYPES=1)
 add_definitions(-DWASM_ENABLE_CUSTOM_NAME_SECTION=1)
 add_definitions(-DWASM_ENABLE_AOT_STACK_FRAME=1)
-add_definitions(-DWASM_ENABLE_DUMP_CALL_STACK=1)
+add_definitions(-DWASM_ENABLE_DUMP_CALL_STACK=0)
 add_definitions(-DWASM_ENABLE_PERF_PROFILING=1)
 add_definitions(-DWASM_ENABLE_LOAD_CUSTOM_SECTION=1)
 add_definitions(-DWASM_ENABLE_MODULE_INST_CONTEXT=1)

```

wenyongh, 2024-10-15:
Hi, this is a limitation and we decided not to support it, see:
https://github.com/bytecodealliance/wasm-micro-runtime/issues/2789

Is there any actual use case from you?

sjamesr, 2024-10-15:
My team is evaluating several WASM engines for use in various projects. One of my tasks is to get the spec tests running with our internal CI, and to determine which tests we should skip/ignore.

While I'm not sure why a user would need NULs in their export/import names, it is part of the spec. Should the engine not strive for spec compliance where reasonably possible? This seems like not much additional work.

wenyongh, 2024-10-16:
The string is stored into const string set, and is also emitted into AOT file, and loaded again in aot loader. All the operations assume the string is null terminated, it may be very complex to change their behaviors and it is easy to make things wrong. A workaround that I can think is to change the character `\0` into a non-utf8 character (e.g. 0x80) in the string before storing into the const string set, and in wasm_runtime_lookup_function_raw, do the same change for the input string and use the changed string to call wasm_runtime_lookup_function. The patch is like below, could you check whether it is good to you?
```C
diff --git a/core/iwasm/common/wasm_runtime_common.c b/core/iwasm/common/wasm_runtime_common.c
index cc3b4bb9..3f753763 100644
--- a/core/iwasm/common/wasm_runtime_common.c
+++ b/core/iwasm/common/wasm_runtime_common.c
@@ -2277,6 +2277,32 @@ wasm_runtime_lookup_function(WASMModuleInstanceCommon *const module_inst,
     return NULL;
 }
 
+WASMFunctionInstanceCommon *
+wasm_runtime_lookup_function_raw(WASMModuleInstanceCommon *const module_inst,
+                                 const char *name, uint32 len)
+{
+    WASMFunctionInstanceCommon *func;
+    char *name_c_str;
+    uint32 i;
+
+    if (!(name_c_str = wasm_runtime_malloc(len + 1))) {
+        return NULL;
+    }
+
+    bh_memcpy_s(name_c_str, len, name, len);
+    name_c_str[len] = '\0';
+
+    for (i = 0; i < len; i++) {
+        if (name_c_str[i] == '\0')
+            name_c_str[i] = 0x80;
+    }
+
+    func = wasm_runtime_lookup_function(module_inst, name_c_str);
+
+    wasm_runtime_free(name_c_str);
+    return func;
+}
+
 uint32
 wasm_func_get_param_count(WASMFunctionInstanceCommon *const func_inst,
                           WASMModuleInstanceCommon *const module_inst)
diff --git a/core/iwasm/common/wasm_runtime_common.h b/core/iwasm/common/wasm_runtime_common.h
index fb2c7940..58f869b4 100644
--- a/core/iwasm/common/wasm_runtime_common.h
+++ b/core/iwasm/common/wasm_runtime_common.h
@@ -616,6 +616,10 @@ WASM_RUNTIME_API_EXTERN WASMFunctionInstanceCommon *
 wasm_runtime_lookup_function(WASMModuleInstanceCommon *const module_inst,
                              const char *name);
 
+WASM_RUNTIME_API_EXTERN WASMFunctionInstanceCommon *
+wasm_runtime_lookup_function_raw(WASMModuleInstanceCommon *const module_inst,
+                                 const char *name, uint32 len);
+
 /* Internal API */
 WASMFuncType *
 wasm_runtime_get_function_type(const WASMFunctionInstanceCommon *function,
diff --git a/core/iwasm/include/wasm_export.h b/core/iwasm/include/wasm_export.h
index dfb6b802..aaab591e 100644
--- a/core/iwasm/include/wasm_export.h
+++ b/core/iwasm/include/wasm_export.h
@@ -793,6 +793,10 @@ WASM_RUNTIME_API_EXTERN wasm_function_inst_t
 wasm_runtime_lookup_function(const wasm_module_inst_t module_inst,
                              const char *name);
 
+wasm_function_inst_t
+wasm_runtime_lookup_function_raw(const wasm_module_inst_t module_inst,
+                                 const char *name, uint32 len);
+
 /**
  * Get parameter count of the function instance
  *
diff --git a/core/iwasm/interpreter/wasm_runtime.c b/core/iwasm/interpreter/wasm_runtime.c
index f6d5e103..47883a7b 100644
--- a/core/iwasm/interpreter/wasm_runtime.c
+++ b/core/iwasm/interpreter/wasm_runtime.c
@@ -4870,12 +4870,7 @@ wasm_check_utf8_str(const uint8 *str, uint32 len)
     while (p < p_end) {
         chr = *p;
 
-        if (chr == 0) {
-            LOG_WARNING(
-                "LIMITATION: a string which contains '\\00' is unsupported");
-            return false;
-        }
-        else if (chr < 0x80) {
+        if (chr < 0x80) {
             p++;
         }
         else if (chr >= 0xC2 && chr <= 0xDF && p + 1 < p_end) {
@@ -4936,6 +4931,7 @@ wasm_const_str_list_insert(const uint8 *str, uint32 len, WASMModule *module,
                            uint32 error_buf_size)
 {
     StringNode *node, *node_next;
+    uint32 i;
 
     if (!wasm_check_utf8_str(str, len)) {
         set_error_buf(error_buf, error_buf_size, "invalid UTF-8 encoding");
@@ -4946,6 +4942,11 @@ wasm_const_str_list_insert(const uint8 *str, uint32 len, WASMModule *module,
         return "";
     }
     else if (is_load_from_file_buf) {
+        for (i = 0; i < len; i++) {
+            if (str[i] == '\0')
+                ((char *)str)[i] = 0x80;
+        }
+
         /* As the file buffer can be referred to after loading, we use
            the previous byte of leb encoded size to adjust the string:
            move string 1 byte backward and then append '\0' */
@@ -4977,6 +4978,11 @@ wasm_const_str_list_insert(const uint8 *str, uint32 len, WASMModule *module,
     bh_memcpy_s(node->str, len + 1, str, len);
     node->str[len] = '\0';
 
+    for (i = 0; i < len; i++) {
+        if (node->str[i] == '\0')
+            node->str[i] = 0x80;
+    }
+
     if (!module->const_str_list) {
         /* set as head */
         module->const_str_list = node;

```

yamt, 2024-10-16:
> and in wasm_runtime_lookup_function_raw, do the same change for the input string and use the changed string to call wasm_runtime_lookup_function.

if you mean to expose wasm_runtime_lookup_function_raw to users, i guess it should perform utf-8 validation on the user-given name before escaping 0.
also, it's better to make it possible for users to distinguish "not found" and "memory allocation failure".

wenyongh, 2024-10-16:
> if you mean to expose wasm_runtime_lookup_function_raw to users, i guess it should perform utf-8 validation on the user-given name before escaping 0. also, it's better to make it possible for users [...]

Totally agree, thanks.