wasm-micro-runtime/wasm-micro-runtime #3870: Do you have any plans to support fast JIT on AARCH64?

ChDC, 2024-10-22:
I want to use WAMR on Android and reduce memory usage by trying the FAST JIT mode. However, it does not support the AARCH64 architecture. I would like to ask if there are any plans to support it in the near future.
If you could also provide information on the costs and challenges associated with supporting Fast JIT on AARCH64, that would be greatly appreciated. Thank you!

TianlongLiang, 2024-10-23:
I don't think anyone is working on this right now so it probably won't be available in the near future.
Suppose WAMR were to support fast-jit on new architecture. The cost is indeed nontrivial because the modification in the Fast JIT engine will be adding the entire code generation process for the AARCH64. As you can see in [code generation process for x86](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/fast-jit/cg/x86-64/jit_codegen_x86_64.cpp), it quite a work and require familiarity with the target ISA. And I don't think the majority of work can be reused (logically maybe, but code-wise it's probably not) to support new architecture. Probably the best approach is to abstract the code generation process and choose implementation based on target architecture if WAMR were to support multiple architectures, but it's quite a lot of work.

ChDC, 2024-10-23:
Thank you for your reply, and I wish you all the best.

closed 2024-10-23 by ChDC