wasm-micro-runtime/wasm-micro-runtime #3875: Add xtensa to wamrc build CI

AlixANNERAUD, 2024-10-23:
## Feature

- Add Xtensa to wamrc build CI.

## Benefit

- Avoid long CI runs on projects that depend on WAMR and clunky dependencies.

## Implementation

Use build_llvm_xtensa.sh instead of the regular build_llvm.sh in the CI.

## Alternatives

None provided.

lum1n0us, 2024-10-23:
Not sure but is this https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/.github/workflows/spec_test_on_nuttx.yml what you are looking for?

AlixANNERAUD, 2024-10-23:
Ok then, is it possible to add the build binaries of `wamrc` (Xtensa version) to the WAMR release.?

lum1n0us, 2024-10-31:
If we're talking about putting in a new backend for Xtensa in wamrc, we can certainly do it.

lum1n0us, 2024-12-25:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3911