wasm-micro-runtime/wasm-micro-runtime #3904: CI: test-wamr-ide is not stable

lum1n0us, 2024-11-12:
Last pass https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/11694609446/job/32568704604
First failure https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/11717281799/job/32636759641

Most like:

```
[3389:1107/054219.675503:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")

  Unit Tests
    ✔ DebugConfigurationProvider init commands
    ✔ DebugConfigurationProvider resolve configuration
  Inegration Tests
[3389:1107/054221.18[69](https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/11717281799/job/32636759641#step:12:70)01:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[3389:1107/054222.303605:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
    1) Rust formatters
  2 passing (8s)
  1 failing
  1) Inegration Tests
       Rust formatters:
     AssertionError: Could not connect to debug adapter
  	at Context.<anonymous> (out/test/suite/extension.test.js:140:31)
  	at Generator.throw (<anonymous>)
  	at rejected (out/test/suite/extension.test.js:10:65)

Error: 1 tests failed.
	at /home/runner/work/wasm-micro-runtime/wasm-micro-runtime/test-tools/wamr-ide/VSCode-Extension/out/test/suite/index.js:28:27
	at done (/home/runner/work/wasm-micro-runtime/wasm-micro-runtime/test-tools/wamr-ide/VSCode-Extension/node_modules/mocha/lib/mocha.js:1028:7)
Extension host test runner error Error: 1 tests failed.
	at /home/runner/work/wasm-micro-runtime/wasm-micro-runtime/test-tools/wamr-ide/VSCode-Extension/out/test/suite/index.js:28:27
	at done (/home/runner/work/wasm-micro-runtime/wasm-micro-runtime/test-tools/wamr-ide/VSCode-Extension/node_modules/mocha/lib/mocha.js:1028:7)
Asking native host service to exit with code 1.
``

lum1n0us, 2024-11-12:
``` ts
test('Rust formatters', async function () {
    // ...
    try {
        await vscode.debug.startDebugging(undefined, config);
    } catch (e) {
        assert.fail('Could not connect to debug adapter');
    }
    // ...
}
```

TianlongLiang, 2024-11-19:
> ```ts
> test('Rust formatters', async function () {
>     // ...
>     try {
>         await vscode.debug.startDebugging(undefined, config);
>     } catch (e) {
>         assert.fail('Could not connect to debug adapter');
>     }
>     // ...
> }
> ```

this error is due to the recent version of the rust toolchain, which will emit ref types opcodes in wasm file, need to enable this feature in the iwasm build.

I also find that the vector format checks fails recently:
```
      - (5) vec![{...}, {...}, {...}, {...}, {...}, ...]
      + (5) vec![1, 2, 3, 4, 12]
```

And previously deque failed the assertion too:
```
      -alloc::collections::vec_deque::VecDeque<int, alloc::alloc::Global> @ 0xfff3c
      + (5) VecDeque[1, 2, 3, 4, 5]
```

I disable those two checks for now. However, I think the ideal fix is to modify the `rust.py` to parse the current format correctly.

TianlongLiang, 2024-11-19:
I also take a look at what it's like in lldb(without import the rust.py formatted):
```
(lldb) frame variable vector
(alloc::vec::Vec<int, alloc::alloc::Global>) vector = {
  buf = {
    inner = {
      ptr = {
        pointer = (pointer = "\U00000001")
        _marker = {}
      }
      cap = (__0 = 8)
      alloc = {}
    }
    _marker = {}
  }
  len = 5
}
```

When I try to dereference it, it's all 0 instead of 1, 2, 3, 4, 5:

```
(lldb) frame variable vector.buf.inner.ptr.pointer
(core::ptr::non_null::NonNull<unsigned char>) vector.buf.inner.ptr.pointer = (pointer = "\U00000001")
(lldb) memory read --size 4 --format d --count 5 0x00000001
0x00000001: 0
0x00000005: 0
0x00000009: 0
0x0000000d: 0
0x00000011: 0
```

lum1n0us, 2024-11-20:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3913

lum1n0us, 2024-11-22:
It's back.

https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/11935315916

lum1n0us, 2024-11-24:
And it runs well again. 

https://github.com/bytecodealliance/wasm-micro-runtime/actions/workflows/compilation_on_android_ubuntu.yml

closed 2024-11-24 by lum1n0us

lum1n0us, 2024-12-06:
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/12182574392/job/34017735221