wasm-micro-runtime/wasm-micro-runtime #3927: Instruction metering

AlixANNERAUD, 2024-11-23:
# Feature

Similar to the Wasmer runtime, it would be beneficial to implement instruction metering, allowing control to be handed back to the caller after executing a specified number of instructions.

# Benefit

This would enable a cooperative scheduler to manage how WebAssembly (Wasm) code is executed, preventing system overload.

# Implementation

Introduce a function like `wasm_func_call_with_metering` that returns an after a given number of executed instruction, a boolean indicating whether the function reached its end.

# Alternatives

None identified.

lum1n0us, 2024-11-26:
Link similar things together. https://github.com/bytecodealliance/wasm-micro-runtime/issues/2163

AlixANNERAUD, 2025-02-20:
Here’s your text with corrections and improvements in clarity and flow:  

---

Passing the number of instructions to be executed seems to be the right approach. However, after inspecting the call stack of `wasm_runtime_call_wasm`, we need to propagate this "instruction count" down to the following functions:  

- **For the interpreter:**  
  1. `wasm_call_function`  
  2. `interp_call_wasm` (a.k.a. `wasm_interp_call_wasm` / `call_wasm_with_hw_bound_check`)  
  3. `wasm_interp_call_func_bytecode`  

  In this final function, we can use the instruction count to exit the interpreter loop after executing *N* instructions and return (by propagating an additional variable to the initial call). However, I’m not exactly sure where the loop condition is checked within this `goto` mess.  

- **For AOT:**  
  I'm not entirely sure, but I believe this concerns the compilation process, where a return state must be added every *N* instructions.  

---

This version improves grammar, word choice, and readability while maintaining technical accuracy. Let me know if you'd like further refinements! 🚀

AlixANNERAUD, 2025-02-20:
Passing the number of instructions to be executed seems to be the right approach. However, after inspecting the call stack of `wasm_runtime_call_wasm`, we need to propagate this "instruction count" down to the following functions:  

- **For the interpreter:**  
  1. `wasm_call_function`  
  2. `interp_call_wasm` (a.k.a. `wasm_interp_call_wasm` / `call_wasm_with_hw_bound_check`)  
  3. `wasm_interp_call_func_bytecode`  

In this final function, we can use the instruction count to exit the interpreter loop after executing *N* instructions and return (by propagating an additional variable to the initial call). However, I’m not exactly sure where the loop condition is checked within this `goto` mess.  

- **For AOT:**  
I'm not entirely sure, but I believe this concerns the compilation process, where a return state must be added every *N* instructions.