wasm-micro-runtime/wasm-micro-runtime #3963: Incorrect Handling of SIMD v128 Return Value

Q1IQ, 2024-12-17:
### Subject of the issue

WasmEdge fails to handle v128 return values correctly, misinterpreting them as instructions and parameters.

### Test case
[program1.wasm.txt](https://github.com/user-attachments/files/18164148/program1.wasm.txt)

### Your environment
* os: Ubuntu 20.04 
* cpu: amd64
* WAMR version: iwasm 1.2.3
* command:  `iwasm --fast-jit ./program.wasm` and `iwasm --interp ./program.wasm`  yield the same result.

### Steps to reproduce

Run the following wasm module:  
```
(module
  (memory 1)
  (func $return_v128 (result v128)
    (v128.const i32x4 0x00000004 0x34567890 0x66666666 0x77777777)
 )
  (export "main" (func $return_v128))
)
```
Observed output:
```
WASM module load failed: alignment must not be larger than natural
```
Run the following wasm module:  
```
(module
  (memory 1)
  (func $return_v128 (result v128)
    (v128.const i32x4 0x00000300 0x34567890 0x66666666 0x77777777)
 )
  (export "main" (func $return_v128))
)
```
Observed output:
```
WASM module load failed: type mismatch: expect data but stack was empty
```

### Expected behavior

The `v128` value should be returned correctly, like it can correctly process i32.
```
(module
  (memory 1)
  (func $func1 (result i32)
    i32.const 1234
 )
  (export "main" (func $func1))
)
```
```
iwasm --interp  -f main program.wasm
0x1234:i32
```

lum1n0us, 2024-12-17:
@Q1IQ SIMD on interp is currently in development. You may want to try again using the development branch. [*dev/simd_for_interp*](https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/simd_for_interp)

@Zzzabiyaka @loganek Just FYI.

Zzzabiyaka, 2024-12-17:
Hi
Just fyi I would expect this branch to be merge-ready in January

lum1n0us, 2024-12-18:
> iwasm --fast-jit ./program.wasm

SIMD is not supported on the fast-jit. Only llvm-jit and aot support SIMD opcodes.

Q1IQ, 2024-12-18:
I tested using the `--llvm-jit` option, and the result remained the same. Additionally, even if SIMD is not supported, it would be better if SIMD values were not interpreted as instructions, as this might cause errors or lead to unexpected behavior.

Q1IQ, 2024-12-18:
interp should at least output a hint like "unsupported opcode" similar to fast-jit, instead of directly segmentation fault.
```
$ iwasm --fast-jit  -f  main ./program.wasm
fast jit compilation failed: Error: unsupported opcode
Exception: failed to compile fast jit function
fast jit compilation failed: Error: unsupported opcode
failed to compile fast jit function 0
$ iwasm --interp  -f main./program.wasm
Segmentation fault (core dumped)
```

Zzzabiyaka, 2024-12-18:
> interp should at least output a hint like "unsupported opcode" similar to fast-jit, instead of directly segmentation fault.
>
> ```
>
> $ iwasm --fast-jit  -f  main ./program.wasm
>
> fast jit compilation [...]

I think for fast interp it might not be needed right now given we expect to deliver simd there in about a month or so

For classic interp maybe it'd be helpful

lum1n0us, 2024-12-29:
Please re-try with latest commit, like 04f1071f1c08b47d734acf0e74511bfeef7c5568. 

if with 

```wat
(module
  (memory 1)
  (func $return_v128 (result v128)
    (v128.const i32x4 0x00000004 0x34567890 0x66666666 0x77777777)
 )
  (export "main" (func $return_v128))
)
```

->

``` bash
$ /workspaces/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --interp  -f main case1.wasm 
Exception: unsupported opcode
```

if with

``` wat
(module
  (memory 1)
  (func $func1 (result i32)
    i32.const 1234
 )
  (export "main" (func $func1))
)
```

->

``` bash
$ /workspaces/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --interp  -f main case2.wasm 
Exception: unsupported opcode
```