wasm-micro-runtime/wasm-micro-runtime #3964: Missing customSections function in wasm_c_api.h

AY268, 2024-12-18:
I want to use the WebAssembly.Module.customSections function, but I couldn't find it in wasm_c_api.h. 
I wonder, if I really want to use customSections in WAMR, how can I achieve this? 
Also, does WAMR plan to include this function in future updates?

lum1n0us, 2024-12-22:
Compared to this definition, there are similar APIs. `wasm_runtime_get_custom_section()` at https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/include/wasm_export.h#L2013C1-L2015C66

The main difference between the two is that the JS APIs return a copy, while the C APIs return a reference.

AY268, 2024-12-24:
> Compared to this definition, there are similar APIs. `wasm_runtime_get_custom_section()` at [...]

Thank you for your help, it really works!

AY268, 2024-12-24:
@lum1n0us 
Hello, I have another question here. I tried to import memory/table to a wasm instance, but I found that WAMR does not copy the data in the memory/table. I wonder if you have any ideas about it. Really appreciate your help.