wasm-micro-runtime/wasm-micro-runtime #3966: First WAMR function call takes significantly longer time due to the OS hardware bound check

sjamesr, 2024-12-19:
Engine: WAMR Fast Intepreter

On the first call to `wasm_runtime_call_wasm` with HW bounds checking enabled, WAMR ultimately ends up calling `call_wasm_with_hw_bound_check`, which has this:

```
uint8 *
os_thread_get_stack_boundary()
{
  /* ... */
#ifdef __linux__
    if (pthread_getattr_np(self, &attr) == 0) {
```

Unfortunately this call to `pthread_getattr_np` is very slow on the main thread, an issue that has been noted in other projects (e.g.  https://github.com/golang/go/issues/68587)

In a particular environment we have, the first call takes 9ms, the second and subsequent calls take 0.5ms.

sjamesr, 2024-12-19:
I think the problem is that [this check in glibc pthreads](https://codebrowser.dev/glibc/glibc/nptl/pthread_getattr_np.c.html#62) returns `false`, so `/proc/self/maps` has to be read and parsed.

sjamesr, 2024-12-19:
Would the WAMR project be amenable to moving the cost of stack boundary computation earlier in the process, e.g. in exec env creation? This is done in https://github.com/bytecodealliance/wasm-micro-runtime/pull/3967.

sjamesr, 2024-12-19:
Just changed the title, because this happens with hardware bounds checking disabled as well.

yamt, 2024-12-20:
have you (or someone) reported the issue to glibc?

sjamesr, 2024-12-20:
There is a discussion of a proposed fix here: https://sourceware.org/pipermail/libc-alpha/2022-September/141932.html, in response to https://internals.rust-lang.org/t/who-is-doing-read-proc-self-maps-1024-at-startup/17348/9. So the issue is known, but I don't think a solution has been agreed and there hasn't been any discussion for a while.

sjamesr, 2025-05-01:
Just a follow-up: with hardware bounds checking disabled, we still call `os_thread_get_stack_boundary`, which is then used in the interpreter in all the call paths, e.g. `wasm_interp_call_wasm` calls `wasm_runtime_detect_native_stack_overflow`.

One thing I'm trying to understand is why there appear to be two bounds checks for every call. Right after `wasm_interp_call_wasm` does `detect_native_stack_overflow`, it calls ALLOC_FRAME, which itself calls `wasm_exec_env_alloc_wasm_frame`, which does a WASM stack overflow.

Can you help me understand why two stack bounds checks are necessary? One checks `exec_env.wasm_stack` bounds, the other checks `exec_env->native_stack_boundary`. The latter check requires that expensive call to `pthread_getattr_np`, which is why I'm interested to see if it can be skipped if hardware bounds checking is disabled.

Thank you!

lum1n0us, 2025-05-06:
Please refer to it to get some background and history. https://github.com/bytecodealliance/wasm-micro-runtime/issues/3325

In a nutshell, WAMR is not only trying to protect against stack overflow from the execution of Waasm functions but also from the execution of host functions imported by Wasm modules.

sjamesr, 2025-05-08:
Our application runs on android, which itself is installing guard pages in its pthread implementation. Do you think it would be safe to rely on this mechanism to detect an overflow, thus avoiding the need to call this expensive `pthread_getattr_np`?

lum1n0us, 2025-05-08:
Definitely worth giving it a try.