wasm-micro-runtime/wasm-micro-runtime #3977: WASI-SDK 25 call_indrect doesn't work on Darwin

woodsmc, 2024-12-19:
Thanks for filing a bug or defect report! Please fill out the TODOs below.

### Subject of the issue

WASI SDK 25 generates a call indirect which uses a ref type format. It doesn't parse correctly on the default build of iwasm for Darwin. Executing the generated wasm with the default iwasm for Darwin results in : ``WASM module load failed: zero byte expected` 

### Test case

Upload the related wasm file, wast file or the source files if you can.
```c
#define WASM_EXPORT(A) __attribute__((export_name(A)))

typedef int(*TFunction)(int);

int function(int x) {
	x *= 25;
	return x;
}

int work(int x, TFunction ptr) {
	return (*ptr)(x);
}

WASM_EXPORT("test") int test(int x) {
	return work(x, &function);
}
```
Compile this with wasi-sdk-25 and attempt to execute with iwasm results in an error message - ``WASM module load failed: zero byte expected` 

### Your environment

* Host OS: MACOS
* WAMR version, platform, cpu architecture, running mode, etc.

### Steps to reproduce

Tell us how to reproduce this bug or defect.

Build the code above with wasi-sdk-25, attempt to run with default iwasm on darwin. Receive error message as below

```
iwasm -f test ./indirect.wasm 25
```
And the error message ``WASM module load failed: zero byte expected` is displayed.

### Expected behavior

Tell us what should happen: The function should execute

### Actual behavior

Tell us what happens instead: WAMR displays error message instead.

### Extra Info

Anything else you'd like to add?

woodsmc, 2024-12-19:
I think turning on reference types resolves this issue. This PR changes the default CMake file for Darwin to do this. https://github.com/bytecodealliance/wasm-micro-runtime/pull/3978

It appeared to be turned on by default on the linux build at the moment, just not on Darwin.

loganek, 2024-12-20:
Hi @woodsmc I think it's reasonable to enable ref types by default. Having said that, I think we could do a better job in terms of error reporting and have clear message when iwasm was compiled with ref types disabled but the binary was compiled with ref types enabled. I can make the change once I'm back from holiday (January next year) so let's keep the issue open until the error handling is improved so I don't forget about it :)

woodsmc, 2024-12-20:
@loganek That's a great idea, if I get the chance over the holidays, I might try to add that concept to the PR - fir this specific case anyway. Or - are you thinking of something more compiler flags, error specific. - Generating a table of the compiled flag options, and using this to allow the WAMR runtime to report how it's compiled, and use the same structure to aid with error condition diagnosis and messaging to the user ?

loganek, 2024-12-20:
@woodsmc AFAIK we can get that information (whether ref types are being used in the wasm binary) from the type section - if that's detected, and WAMR was compiled with ref types disabled, we show an error and stop loading.