wasm-micro-runtime/wasm-micro-runtime #3979: Why "Fast JIT currently not supported on Windows"

lxbxl, 2024-12-20:
I would like to understand the reasons why the Fast JIT feature is currently not supported on Windows. 
If there's an opportunity, I would be willing to attempt implementing it myself.

TianlongLiang, 2024-12-23:
This is due to insufficient testing on this platform feature combo. I actually did make it compiled(at least on the MSVC toolchain) on the Windows platform, you can try it too by modifying the [L62-L68 in runtime_lib.cmake](https://github.com/bytecodealliance/wasm-micro-runtime/blob/4cda74ad85e9f18dce0790e86ec8aaef1f76ba5c/build-scripts/runtime_lib.cmake#L62C1-L67C12) and add:

```Cmake
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
```

on Windows iwasm `CMakelists.txt`.

But I am not sure it will work on other toolchains on Windows, and it needs more testing than simply compiling, that's why we disable it for now. It would be great if you could help with the testing, maybe starting by enabling certain spec tests for some running modes for [CI on Windows](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/.github/workflows/compilation_on_windows.yml).

lxbxl, 2024-12-23:
Thank you for your response.

I am currently conducting preliminary research on running a minimal feature set of WAMR in Windows kernel mode. 

As you mentioned, supporting the Fast JIT mode In MSVC requires C++20. I have successfully completed the x64 compilation for Windows user mode; however, the compiled program fails to run successfully. Specifically, it crashes when executing asmjit modules.

By the way, Would you be more comfortable communicating in Chinese? my proficiency in English is limited.

lxbxl, 2024-12-23:
Thank @TianlongLiang for your response.

I am currently conducting preliminary research on running a minimal feature set of WAMR in Windows kernel mode.

As you mentioned, supporting the Fast JIT mode In MSVC requires C++20. I have successfully completed the x64 compilation for Windows user mode; however, the compiled program fails to run successfully. Specifically, it crashes when executing asmjit modules.

By the way, Would you be more comfortable communicating in Chinese? my proficiency in English is limited.

TianlongLiang, 2024-12-23:
行啊，现在是报什么错啊？方便的话截图给我看一眼吗？

lxbxl, 2024-12-23:
写了个 simple.c 用 wasi-sdk 编成 simple.wasm。 尝试在用户态的 iwasm.exe 内运行，结果崩了。

simple.c
```c
#include <stdio.h>

int main(int argc, char** argv)
{
	printf("Hello World");
	return 0;
}
```

编译命令：`D:/wasi-sdk-24.0-x86_64-windows/bin/clang.exe -O3 -o simple.wasm simple.c`

运行命令：`./iwasm.exe simple.wasm`

[image]

u.f(exec_env, info, func_idx, target) 处崩溃。

lxbxl, 2024-12-23:
> 行啊，现在是报什么错啊？方便的话截图给我看一眼吗？

前两天 Fast Jit 还测了个宿主程序调用 simple.wasm 中导出函数的模型，初步定位可能是 jit_codegen_init 中的代码导致的崩溃。asmjit 不熟，就没跟下去，环境也没保存。

TianlongLiang, 2024-12-24:
可以试一下编译时加上宏`-DWAMR_BUILD_FAST_JIT_DUMP=1`，看看编译出来了jit code 没有。以及运行时候加上`./iwasm.exe -v=5 simple.wasm` 的看看log。不过如果是 jit_codegen_init的话估计就打印不出来fast jit 的code dump，可能是有些asmjit 的 api 在Windows 上用法不同之类的

lxbxl, 2024-12-24:
> 可以试一下编译时加上宏`-DWAMR_BUILD_FAST_JIT_DUMP=1`，看看编译出来了jit code 没有。以及运行时候加上`./iwasm.exe -v=5 simple.wasm` 的看看log。不过如果是 jit_codegen_init的话估计就打印不出来fast jit 的code dump，可能是有些asmjit 的 api 在Windows 上用法不同之类的

公司决定 FAST JIT 暂不使用，先把 win 内核下的 Fast Interpreter 与 Aot 落地。这个 issue 我先关了，后面有进一步的消息再来回复。

再次感谢你的回复。

closed 2024-12-24 by lxbxl