wasm-micro-runtime/wasm-micro-runtime #3980: Feature Request: Wasm Proposal Details at Compile and Runtime

woodsmc, 2024-12-22:
> > Mentioning the options selected during the build process - and the options not selected would be really great improvement.
> 
> AFAIK some of the options are printed when generating CMake file, but some of them are not. Also, it'd be good to print the enabled features when user calls `iwasm --help` or `iwasm -v`. Would you mind open an issue so we can track the improvement?

I am creating [a new document about which wasm proposals are on-by-default and which are off-by-default](https://github.com/bytecodealliance/wasm-micro-runtime/blob/ee647abfece921f541c0b66f16de4ee76a20cf91/doc/stability_wasm_proposals.md). Are you suggesting we should output something like that during cmake configuration and execution by `iwasm` ? 

Might need to add the library case

_Originally posted by @lum1n0us in https://github.com/bytecodealliance/wasm-micro-runtime/issues/3978#issuecomment-2558411602_

lum1n0us, 2024-12-26:
I have created a https://github.com/bytecodealliance/wasm-micro-runtime/pull/3989 for further discussion.

loganek, 2025-01-02:
> We could suggest to the compiler / tool chain folks that they think about including a custom section with a list of the features enabled by the compiler when the .wasm file was produced. It would be easy to remove this custom section to reduce binary size, but it would provide useful information to the user.

Not sure if that's needed. Is there any feature that's currently not detectable?

lum1n0us, 2025-01-03:
From my perspective, it's a method to inform users about the combination of features they are currently using and to confirm whether it aligns with their expectations, rather than to trace back through the build environment.

loganek, 2025-01-03:
> From my perspective, it's a method to inform users about the combination of features they are currently using and to confirm whether it aligns with their expectations, rather than to trace back through the build environment.

Right, in that case I think it makes sense to have the custom section as suggested by @woodsmc

woodsmc, 2025-01-03:
Yeah, speaking from personal experience, there has been a bunch of times where I've compiled code to .wasm, and discovered it won't execute in the host environment. I have spent hours trying to debug this, only to discover that the .wasm needed features which were not turned on in the host environment. 

Anything that helps identify this would be amazing. - The reason why I'm suggesting to reach out to the WASI-SDK folks is that I think they are already inserting a lot of additional information in the .wasm file already. This also seems to be growing. If you check the compiler output for WASI-SDK 21 and compare it to WASI-SDK 25 you can see that the same code results in a much larger .wasm file. For example, a simple "Hello World" example compiles with (-O3) to 4.5kb using wasi-sdk 21, but compiles to a much larger 49kb using wasi-sdk 25. It looks like it's adding 9 custom sections with a lot of debug information.

yamt, 2025-01-06:
> We could suggest to the compiler / tool chain folks that they think about including a custom section with a list of the features enabled by the compiler when the `.wasm` file was produced. It would [...]

you can check the existing target_features section if you want.

yamt, 2025-01-06:
> If you check the compiler output for WASI-SDK 21 and compare it to WASI-SDK 25 you can see that the same code results in a much larger .wasm file. For example, a simple "Hello World" example [...]

it's mainly because wasi-sdk has turned on the debug info.
it's recommended to strip your binary if the size matters.

loganek, 2025-01-06:
> Yeah, speaking from personal experience, there has been a bunch of times where I've compiled code to .wasm, and discovered it won't execute in the host environment. 

Could you give some more-concrete examples? I think in most (all?) cases we should be able to provide a meaningful error message to the user, even if the "features" section does not exist (e.g. similarly to what we did in https://github.com/bytecodealliance/wasm-micro-runtime/pull/4000 for the reference types)