wasm-micro-runtime/wasm-micro-runtime #3988: Invoke other wasm function in native api: "Exception: invalid exec env"

Jiax-cn, 2024-12-26:
I'm trying to invoke a wasm function (of another instance) in my native api, but failed with "Exception: invalid exec env". Is it impossible to achieve this without setting WAMR_DISABLE_HW_BOUND_CHECK? Any guidance would be appreciated.

lum1n0us, 2024-12-26:
It's advisable not to use `WAMR_DISABLE_HW_BOUND_CHECK` to switch to software bounds checking unless the platform has poor support for `signals` and `jmp`.

For reference, please look at the examples in samples/basic or samples/spawn-thread, depending on your specific use case.

If that's acceptable, please share your embedded code for discussion.

Jiax-cn, 2024-12-26:
Thanks for your response. Here is a simplified version of the code I’m working with:

hello.wasm
```
(import "my_module" "world" (func $world_func))
(func $_start 
  ;; printf("hello ")
  call $world_func
)
```

world.wasm
```
(func $_start 
  ;; printf("world!\n")
)
```

host_app
```
int main() {
    ...
    hello_module = wasm_runtime_load(hello_wasm ...);
    hello_module_inst = wasm_runtime_instantiate(hello_module, ...);
    hello_func = wasm_runtime_lookup_function(hello_module_inst, "_start");
    hello_exec_env = wasm_runtime_create_exec_env(hello_module_inst, ...);

    world_module = wasm_runtime_load(world_wasm ...);
    world_module_inst = wasm_runtime_instantiate(world_module, ...);
    world_func = wasm_runtime_lookup_function(world_module_inst, "_start");
    world_exec_env = wasm_runtime_create_exec_env(world_module_inst, ...);

    wasm_runtime_call_wasm(hello_exec_env, hello_func, ...);

    ...
}
```

native api wrapper
```
static void my_module_world_wrapper(...) {
    if (!wasm_runtime_call_wasm(world_exec_env, world_func, ...)) {
        // Exception: invalid exec env
        os_printf("%s\n", wasm_runtime_get_exception(world_module_inst));
    }
}
```

expected result:
```
hello world!
```

but result:
```
hello

#00: 0x0000 - _start

Exception: invalid exec env
```

lum1n0us, 2025-01-20:
Ah, I understand now. In that situation, you would require the multi-module feature to complete the task(`-DWAMR_BUILD_MULTI_MODULE=1`). To put it simply, since hello.wasm depends on world.wasm, they should be connected through import and export. Consequently, WAMR will recognize this connection while loading hello.wasm and will load world.wasm automatically. This process is what we refer to as loading-linking.

You can find an example that addresses this type of problem [here](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/multi-module). It serves as a reference for resolving similar cases.