wasm-micro-runtime/wasm-micro-runtime #3997: Error: "Exception: create singleton exec_env failed" on ESP32 and Rust

SeaRoll, 2024-12-31:
Hi!

I am using ESP32 on Rust via `esp-idf-sys` and WAMR linking component as such:
```toml
[[package.metadata.esp-idf-sys.extra_components]]
component_dirs = ["/esp/wasm-micro-runtime/build-scripts/esp-idf"]
bindings_header = "/esp/wasm-micro-runtime/core/iwasm/include/wasm_export.h"
bindings_module = "wamr"
```

And the code looking something like this for main.rs

```rust
use std::ffi::CStr;

use esp_idf_sys::wamr::{
    wasm_application_execute_main, wasm_runtime_full_init, wasm_runtime_get_exception,
    wasm_runtime_instantiate, wasm_runtime_load, RuntimeInitArgs,
};

fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Initializing WAMR runtime");

    // RuntimeInitArgs init_args;
    // /* configure memory allocation */
    // memset(&init_args, 0, sizeof(RuntimeInitArgs));
    // init_args.mem_alloc_type = Alloc_With_Allocator;
    // init_args.mem_alloc_option.allocator.malloc_func = (void *)os_malloc;
    // init_args.mem_alloc_option.allocator.realloc_func = (void *)os_realloc;
    // init_args.mem_alloc_option.allocator.free_func = (void *)os_free;

    let mut init_args = RuntimeInitArgs::default();
    init_args.mem_alloc_type = 1;
    init_args.mem_alloc_option.allocator.malloc_func = esp_idf_svc::sys::malloc as _;
    init_args.mem_alloc_option.allocator.realloc_func = esp_idf_svc::sys::realloc as _;
    init_args.mem_alloc_option.allocator.free_func = esp_idf_svc::sys::free as _;

    // Initialize the WAMR runtime
    unsafe {
        wasm_runtime_full_init(&mut init_args);
    }

    let mut wasm_module = include_bytes!("main.wasm").to_vec();
    log::info!("Wasm file bytes loaded! {:?}", wasm_module.len());
    let mut error_buf = [0i8; 128];

    let module_handle = unsafe {
        wasm_runtime_load(
            wasm_module.as_mut_ptr(),
            wasm_module.len() as u32,
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        )
    };

    if module_handle.is_null() {
        log::error!("Failed to load Wasm module: {:?}", error_buf);
        return;
    }

    log::info!("Wasm module loaded!");
    // Instantiate the module
    let instance_handle = unsafe {
        wasm_runtime_instantiate(
            module_handle,
            4 * 1024 * 1024, // stack size
            4 * 1024 * 1024, // heap size
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        )
    };

    if instance_handle.is_null() {
        log::error!("Failed to instantiate Wasm module: {:?}", error_buf);
        let c_str = unsafe { CStr::from_ptr(error_buf.as_ptr()) };
        log::error!("Error: {:?}", c_str);
        return;
    }

    log::info!("Wasm module loaded and instantiated!");

    // run main
    let result = unsafe {
        wasm_application_execute_main(instance_handle, 0, 0usize as _);
    };
    let exception = unsafe { wasm_runtime_get_exception(instance_handle) };

    if !exception.is_null() {
        log::error!("Exception: {:?}", exception);
        let c_str = unsafe { CStr::from_ptr(exception) };
        log::error!("Error: {:?}", c_str.to_string_lossy().into_owned());
        return;
    }

    log::info!("Result: {:?}", result);
    log::info!("Runtime done running!");
}
```

I am although getting from my esp32s3 device:

```sh
I (595) hello_idf: Wasm module loaded!
I (595) hello_idf: Wasm module loaded and instantiated!
E (595) hello_idf: Exception: 0x3fcbadf4
E (595) hello_idf: Error: "Exception: create singleton exec_env failed"
```

Any thoughts?

lum1n0us, 2025-01-04:
Just some thoughts.

- check the return value of `wasm_runtime_full_init(&mut init_args)`
- compilation options used when building the runtime library

SeaRoll, 2025-01-06:
> Just some thoughts.
> 
> * check the return value of `wasm_runtime_full_init(&mut init_args)`
> * compilation options used when building the runtime library

I will take a look into this, thank you!

SeaRoll, 2025-01-11:
Tried it out and got a code like this:

```rust
use std::ffi::CStr;

use esp_idf_sys::wamr::{
    wasm_application_execute_main, wasm_runtime_deinstantiate, wasm_runtime_destroy,
    wasm_runtime_full_init, wasm_runtime_get_exception, wasm_runtime_instantiate,
    wasm_runtime_load, wasm_runtime_unload, RuntimeInitArgs,
};

fn iwasm_main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Initializing WAMR runtime");

    let mut init_args = RuntimeInitArgs::default();
    init_args.mem_alloc_type = 1;
    // init_args.mem_alloc_option.pool.heap_size = 1024 * 1024 * 8; // 8 MB heap size
    init_args.mem_alloc_option.allocator.malloc_func = esp_idf_svc::sys::malloc as *mut _;
    init_args.mem_alloc_option.allocator.realloc_func = esp_idf_svc::sys::realloc as *mut _;
    init_args.mem_alloc_option.allocator.free_func = esp_idf_svc::sys::free as *mut _;

    // Initialize the WAMR runtime
    let success: bool = unsafe { wasm_runtime_full_init(&mut init_args) };
    if !success {
        log::error!("Failed to initialize WAMR runtime");
        return;
    }

    let mut wasm_module = include_bytes!("output.wasm").to_vec();
    log::info!("Wasm file bytes loaded! {:?}", wasm_module.len());
    let mut error_buf = [0i8; 128];

    let module_handle = unsafe {
        wasm_runtime_load(
            wasm_module.as_mut_ptr(),
            wasm_module.len() as u32,
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        )
    };

    if module_handle.is_null() {
        let c_str = unsafe { CStr::from_ptr(error_buf.as_ptr()) };
        log::error!("Failed to load Wasm module: {:?}", c_str);
        return;
    }

    log::info!("Wasm module loaded!");
    // Instantiate the module
    let instance_handle = unsafe {
        wasm_runtime_instantiate(
            module_handle,
            16 * 1024, // stack size
            16 * 1024, // heap size
            error_buf.as_mut_ptr(),
            error_buf.len() as u32,
        )
    };

    if instance_handle.is_null() {
        let c_str = unsafe { CStr::from_ptr(error_buf.as_ptr()) };
        log::error!("Failed to instantiate Wasm module: {:?}", c_str);
        return;
    }

    log::info!("Wasm module loaded and instantiated!");

    // run main
    let argv: *mut *mut i8 = std::ptr::null_mut();
    let success = unsafe { wasm_application_execute_main(instance_handle, 0, argv) };

    if !success {
        log::error!("Failed to execute Wasm module");
        let exception = unsafe { wasm_runtime_get_exception(instance_handle) };
        if !exception.is_null() {
            let c_str = unsafe { CStr::from_ptr(exception) };
            log::error!(
                "Error was found during execution: {:?}",
                c_str.to_string_lossy().into_owned()
            );
            return;
        }
    }

    std::thread::sleep(std::time::Duration::from_secs(1));

    // Clean up
    log::info!("Cleaning up runtime");
    unsafe { wasm_runtime_deinstantiate(instance_handle) };
    log::info!("Deinstantiate done!");
    unsafe { wasm_runtime_unload(module_handle) };
    log::info!("Unload done!");
    unsafe { wasm_runtime_destroy() };

    log::info!("Runtime done running!");
}

fn main() {
    // Test running a thread
    let handle = std::thread::spawn(|| {
        iwasm_main();
    });

    if let Err(e) = handle.join() {
        log::error!("Failed to run thread: {:?}", e);
    }
}

```

Now the problem is that I get "Failed to instantiate Wasm module: "WASM module instantiate failed: allocate linear memory failed"". I heard its some allocator problem, but at a stop for now

closed 2025-01-18 by SeaRoll