wasm-micro-runtime/wasm-micro-runtime #4006: Custom-page-size and/or shrink memory

lum1n0us, 2025-01-06:
> https://github.com/WebAssembly/spec/pull/1753#issuecomment-2394262819

The *custom-page-size feature* is a standard proposal that overlaps with the current *shrink_memory implementation based on heap_base in wasm_loader*. The unresolved question is **whether we should retain both features or just maintain one**.

yamt, 2025-01-06:
> Given that shrinking memory is not a standard operation, yet there is consistently a need to reduce memory usage, should we make it controllable and/or configurable? For instance, by using a compilation option and/or a command line option.

it makes sense.

while custom-page-sizes might be able to replace it in future, it's far from ready today.

lum1n0us, 2025-01-09:
#4008 will fix it

closed 2025-01-15 by lum1n0us