wasm-micro-runtime/wasm-micro-runtime #4014: Incorrect handling of table.init with out-of-bounds element segment

Q1IQ, 2025-01-09:
### Subject of the issue

iwasm fails to correctly handle table.init when attempting out-of-bounds table access, resulting in unexpected behavior or a failure to throw the expected exception.

### Test case

[program.wasm.txt](https://github.com/user-attachments/files/18362131/program.wasm.txt)

### Your environment

* OS: Ubuntu 20.04
* CPU: amd64
* WAMR version: iwasm 2.2.0
* Commands:
```
iwasm --interp -f main ./program.wasm
iwasm --fast-jit -f main ./program.wasm
iwasm --llvm-jit -f main ./program.wasm
iwasm --multi-tier-jit -f main./program.wasm
```

### Expected behavior

The WebAssembly runtime should detect the out-of-bounds table access during table.init and report the error. A valid error message should look like:

```
Out of bounds table access
```

TianlongLiang, 2025-01-10:
This is an issue caused by a lack of checking table segment mode(it has to be passive to use by table init), I'm working on it to fix it.

TianlongLiang, 2025-01-10:
Hi, are you using the latest commit? I tested on the latest commit it seems that the behavior is as expected

closed 2025-01-10 by Q1IQ