wasm-micro-runtime/wasm-micro-runtime #4019: Data Persistence Issue with Data Storage Module in Multi-Module Setup Without Host Interaction

fxble, 2025-01-13:
I am encountering an issue with WAMR where data stored in a module does not persist across function calls in a multi-module setup, without any host interaction.

**Setup:**
- **Entry Module:** Host calls entry function, module manages interaction between submodules until finished.
- **Data Storage Submodule:** Provides functions for storing and retrieving data to and from C arrays.
- **Other Submodules:** import and call the storage module's get/set functions to read or write data to the arrays.

**Problem:**
- Modules can call the storage module’s getter and setter functions successfully.
- However, data does not persist between calls. The storage module resets to its initial state, as if a new instance is created for each function call.
- Each data retrieval returns the initial values, rather than the values previously written.
- I would guess the module instance gets deleted after every function call. (?)

**Questions:**
1. Is this behavior expected in WAMR? Or do I probably have some kind of configuration error?
2. Is there any way to achieve such a persistent data module - across multiple submodule interactions and without host intervention?

**Environment:**
- bare-metal microcontroller (STM32, Cortex-M4)
- WAMR 2.2.0 without WASI

**Expected Behavior:**
Data in the submodule should remain persistent and be accessible across all module interactions without reinitialization or host involvement.

**Actual Behavior:**
Data resets to the initial state following each function call.

**BTW:** I know I can implement a similar setup with host-controlled data/memory allocations, but in my experiment I deliberately don't want to do this.

lum1n0us, 2025-01-14:
>  Provides functions for storing and retrieving data to and from C arrays.

Does this imply that the data is actually managed in the native/C world rather than within the WebAssembly world? If that's the case, I would recommend adding logs or setting gdb watchpoints to determine what is resetting the value.

---

I have some theories about what might be happening, but I would need to see the source code to be certain(wat + host embedding).

fxble, 2025-01-14:
In general, I write all module code in C and use _clang/wamrc_ to compile them to _AOT_ files.

---

> Does this imply that the data is actually managed in the native/C world rather than within the WebAssembly world?

From my understanding, the data is managed by the WebAssembly world, because I only create and access arrays inside the data module.

---

**My implementation is as follows:**

The native host code looks like this:
```C
/* Host (main.c) */
wamr_init();
wasm_runtime_register_natives(...);
load_and_register_all_modules(...); // load modules as AOT byte array
wasm_module_inst_t entry_module_inst = wasm_runtime_instantiate(entry_module, ...);
wasm_exec_env_t entry_exec_env = wasm_runtime_create_exec_env(entry_module_inst, ...);
wasm_function_inst_t entry_func_inst = wasm_runtime_lookup_function(entry_module_inst, "entry");

wasm_runtime_call_wasm(entry_exec_env, entry_func_inst, 0, NULL)); // --> call the main modules entry function
```

The host calls the entry function of the main module. This entry function in turn calls imported functions from other modules:
```C
/* Main Module (entry.c)*/
__attribute__((import_module("read")))
__attribute__((import_name("read_something")))
void read_something();

void entry() {
    read_something(); // --> imported function from submodule 1
    do_something(); // --> imported function from submodule 2
    write_something(); // --> imported function from submodule 3
}
```

These submodule functions need access to the array data. That data is managed by another submodule. Therefore, submodules 1, 2 and 3 import the get/set functions from the data storage module:
```C
/* Submodule 1 (read.c) */
__attribute__((import_module("data")))
__attribute__((import_name("set_value")))
void set_value(int value, int index);

__attribute__((export_name("read_something")))
void read_something() {
    /* read something */
    set_value(8, 0); // --> imported function from data storage module
}
```

The data storage module is implemented as seen below:
```C
/* Data Storage Module (data.c) */
int my_array[] = {1, 2, 3};

__attribute__((export_name("get_value")))
int get_value(int index) {
    if(index >= 0 && index < 3) {
        return array[index];
    }
    else {
        return INT_MAX;
    }
}

__attribute__((export_name("set_value")))
void set_value(int value, int index) {
    if(index >= 0 && index < 3) {
        array[index] = value;
    }
}
```

---

**What I forgot to mention:**
I checked the implementation of the data storage module by instantiating it and executing its get/set functions exclusively from the host and it worked. Because the host handles the instantiation and keeps the module instance alive, is that right?

lum1n0us, 2025-01-20:
It's a bit complex here. WAMR creates a new instance for every submodule within the dependency tree, even if multiple submodules are derived from the same module. 

[image]

This implies that read.wasm (from read.c) and write.wasm (from write.c) each utilize an individual instance of data.wasm (from data.c), instead of sharing one instance. Consequently, the `read()` and `write()` functions do not access the same `int[]`, as there are two distinct arrays in the memories of the two separate instances.

[image]

A solution to this would be to consolidate `read()` and `write()` into a single module(.c file). Then, there would be just one `int[]` array in the memory of a single instance.

fxble, 2025-02-03:
Thank you for your clarification. I appreciate it!

---

In my case, I need the read/write operations in each submodules. Therefore, my approach would be to put the array in the main (entry) module. So that every submodule imports/calls read/write from the main module and the array data persists because the main module never dies. 

So far so good. But another problem occurs in this scenario. I need to import functions from the main module into each submodule and vice versa. So there is a cyclic dependency regarding the module registration order:

The main module needs to import every submodule-entry-function and every submodule needs to import the read/write-functions. As the imports can only be resolved if the exporting module gets registered before the importing module, this cyclic redundancy can never be resolved in this situation.

--- 

Is there a better or even recommended way of registering module functions in such a situation?

(FYI: I use the import/export statements like above in my previous code examples.)

fxble, 2025-03-11:
Does anyone know a solution or can give advice on how to resolve such a situation? Or is there something I have overlooked?

lum1n0us, 2025-03-12:
Maybe:

Maintain the `int[]` in the *main* module, and refactor the `read()` and `write()` to use a caller-provided-buffer. Place the new `read()` and `write()` in a submodule, and have other components depend on this new submodule.

fxble, 2025-04-02:
But then I'm going to have the same import/export dependency cycle as mentioned before. The main module calls the entry function from a submodule, and the submodule needs to call the read/write from the main module. So, I have the same problem as before. One module must know a previously registered module in order to import a function and vice versa...at least I run into this kind of error.

Additionally, providing buffers to other modules isn't possible with C, as I can only provide a pointer to an array. And that's not applicable because the modules' memory regions are isolated.

I conclude from this that WAMR's import/export mechanism can only be used in one direction and not bidirectionally, am I right?

fxble, 2025-04-11:
However, I have solved it differently.

I have used module-local arrays in combination with getter and setter functions that are exported and then imported into the main module. This way the main module has to write and/or read data before calling the next module, instead of each module reading/writing data to a dedicated data module.

closed 2025-04-11 by fxble