wasm-micro-runtime/wasm-micro-runtime #4024: Missing UserData in LinearMemory for custom allocator

thomasbechet, 2025-01-14:
### Description

UserData is always NULL for custom allocator when usage is Alloc_For_LinearMemory.

### Test case

Using this configuration :
```
set (WAMR_BUILD_GLOBAL_HEAP_POOL 0)
set (WAMR_BUILD_ALLOC_WITH_USAGE 1)
set (WAMR_BUILD_ALLOC_WITH_USER_DATA 1)
```

With this initialization :
```
init_args.mem_alloc_type                          = Alloc_With_Allocator;
init_args.mem_alloc_option.allocator.malloc_func  = wasm_malloc;
init_args.mem_alloc_option.allocator.realloc_func = wasm_realloc;
init_args.mem_alloc_option.allocator.free_func    = wasm_free;
init_args.mem_alloc_option.allocator.user_data    = vm;
```

With this custom allocator :
```
void * wasm_realloc (mem_alloc_usage_t usage,
              nu_bool_t         full_size_mmapped,
              void             *user,
              void             *p,
              nu_u32_t          n);
void * wasm_malloc (mem_alloc_usage_t usage, void *user, nu_size_t n);
void   wasm_free (mem_alloc_usage_t usage, void *user, void *p);
```

Implement a custom allocator with the given configuration, callbacks.

### Your environment

* Host OS : Linux 
* WAMR version 2.2.0, Linux 

### Steps to reproduce

- Implement custom allocator and print user data pointer.

### Expected behavior

I expect 'vm' (my user data) to be passed to the custom allocator callbacks
in the user parameter.

### Actual behavior

However, it is NULL when usage = Alloc_For_LinearMemory.

### Extra Info

Inspecting the code (wasm_memory.c), it is set to NULL for Alloc_For_LinearMemory allocation
and allocator_user_data for Alloc_For_RuntimeMemory.
Changing the NULL to allocator_user_data fixed the problem.

lum1n0us, 2025-01-21:
FYI: https://github.com/bytecodealliance/wasm-micro-runtime/pull/3316#discussion_r1569918267

I believe we could use `allocator_user_data` as a substitute for NULL in the third parameter when invoking `malloc_func`(`wasm_allocate_linear_memory()`), `realloc_func`(`wasm_enlarge_memory_internal()`), and `free_func`(`wasm_deallocate_linear_memory()`).

@dongsheng28849455 @wenyongh Could you please confirm this for me?

dongsheng28849455, 2025-01-21:
yes, I think it's fine.

PR #4043 (2025-02-06): Update memory allocation functions to use allocator user data
fix #4024
closed 2025-02-06 by lum1n0us