wasm-micro-runtime/wasm-micro-runtime #4028: AOT does not throw error as expected

AY268, 2025-01-16:
I try to test The AOT module of WAMR, and I use the tests in wasm-spec-tests.
I compile it to AOT myself, but when I run this case, I didn't get RuntimeError.

The WebAssembly module is:
(module
  (memory 1)
  (data (i32.const 0) "abcdefghijklmnopqrstuvwxyz")
...
  (func (export "32_good5") (param $i i32) (result i32)
    (i32.load offset=25 align=4 (local.get $i))             ;; 122 'z\0\0\0'
  )
...
)

The test case is:
(assert_trap (invoke "32_good5" (i32.const 65508)) "out of bounds memory access")

I wonder why the AOT module of WAMR does not throw error when I run this test case, and how it could be made to throw one.

TianlongLiang, 2025-01-16:
Hi, could you please provide more details? For example, 
- Your platform and os(mine is x86 and ubuntu 20.04)
- What kind of build and compile flags are you using? I compile iwasm and wamrc with the default compile flag in my local environment, it reports the exception correctly.
  This can be obtained by `cmake -B build` to print the cmake config logs like:
  ```
  -- Build Configurations:
       Build as target X86_64
       CMAKE_BUILD_TYPE Release
       WAMR Interpreter enabled
       WAMR AOT enabled
       WAMR Fast JIT disabled
       WAMR LLVM ORC JIT disabled
       Libc builtin enabled
       Libc WASI enabled
       Fast interpreter enabled
       Multiple modules disabled
       Bulk memory feature enabled
       Wakeup of blocking operations enabled
       SIMD enabled
       Reference types enabled
       GC performance profiling disabled
       Module instance context enabled
       Write linear memory base addr to x86 GS register enabled
       Quick AOT/JIT entries enabled
       AOT intrinsics enabled
       Shrunk memory enabled
  -- Configuring done (0.1s)
  -- Generating done (0.0s)
  ```

Also, you can add `-v=5` option to show more details in iwasm like:
```bash
$IWASM_CMD -v=5 -f 32_good5 test.aot 65508
```

AY268, 2025-01-16:
@TianlongLiang 
This is my build configuration:
```
-- Build Configurations:
     Build as target AARCH64
     CMAKE_BUILD_TYPE RelWithDebInfo
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin enabled
     Libc WASI enabled
     Fast interpreter enabled
     Multiple modules disabled
     Bulk memory feature enabled
     Hardware boundary check disabled
     Wakeup of blocking operations enabled
     Reference types disabled
     GC performance profiling disabled
     Load custom section enabled
     Module instance context enabled
     Quick AOT/JIT entries enabled
     AOT intrinsics enabled
-- Configuring done (0.0s)
-- Generating done (0.0s)
```

When I debug the test, I find that when it runs in the function `invoke_native_internal` (in the file `aot_runtime.c`, the one which does not define `OS_ENABLE_HW_BOUND_CHECK`), it runs through the function `invoke_native`. I think this might be useful information.

Really thanks for your help.

TianlongLiang, 2025-01-16:
From this line: "Hardware boundary check disabled" indicates iwasm disables HW boundary checks and uses software boundary check, so when using wamrc to compile the AOT file, you need to enable software boundary check too:
```bash
$WAMRC --bounds-checks=1 -o test.aot test.wasm
```

AY268, 2025-01-17:
Thanks for your help, it really works!