wasm-micro-runtime/wasm-micro-runtime #4032: A bug in ckecking data count section

erxiaozhou, 2025-01-17:
In the test case, the actual number of data segments (0) does not match the length declared in the data count section (1). According to the specification, "the module is malformed". However, WAMRs in JIT mode, fast JIT mode, and interpreter mode do not raise an exception as expected.

### Test case
[nop_0LM_592_17171016522810388.zip](https://github.com/user-attachments/files/15514361/nop_0LM_592_17171016522810388.zip)

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: 082cfa1c4f8b826f7e22f8e81b2fe76224a3d7bf

### Steps to reproduce
1. build WAMR on fast interpreter mode / classic interpreter mode / JIT mode /  fast JIT mode
2. execute the command
```
iwasm --heap-size=0 -f "to_test" <test_case>
```
### Expected behavior
An exception indicating that "data count and data section have an inconsistent length"
### Actual behavior
No exception

TianlongLiang, 2025-01-20:
Hi, I tried the latest commit and this issue is gone(to be more specific, it was fixed in https://github.com/bytecodealliance/wasm-micro-runtime/pull/3492).

erxiaozhou, 2025-01-20:
Hi, I'm sorry for uploading the wrong test case. I have updated the test case now.Thank you! @TianlongLiang

erxiaozhou, 2025-01-21:
WAMR can not recognize the current test case, which is invalid.