wasm-micro-runtime/wasm-micro-runtime #4036: WAMR behaves differently compared to other runtimes for a particular test case.

teyahb8, 2025-01-19:
### Subject of the issue

WAMR behaves differently compared to other runtimes for a particular test case.

### Test case

Uploaded files.zip contains the wat code, the wasm binary, and the aot file. [files.zip](https://github.com/user-attachments/files/18470322/files.zip)

### Your environment

* WAMR version (2.2.0), platform (Linux), cpu architecture (X86_64), running mode (AOT).

### Steps to reproduce

Run `wat2wasm 53.wat` to produce `53.wasm`
Run `wamrc --bounds-checks=1 -o 53.aot 53.wasm to generate 53.aot`
Run `iwasm -f main 53.aot to manifest the crash`

### Expected behavior

Other runtimes like `wasmtime`, `wasmer`, and `wasmedge` all timeout with this particular test case.

### Actual behavior

Outputs from `wamrc` and `iwasm` are as follows:

wamrc output:
```
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file 53.aot was generated.
```

iwasm output:
```
unhandled SIGSEGV, si_addr: 0x40a36248
```

Can you confirm if this is a bug? Thanks in advance.

TianlongLiang, 2025-01-20:
It's not a bug. I checked your wat file, it contains an infinite loop, and in wamrc(LLVM), the default optimization level(3) will optimize out the infinite loop and cause undefined behavior.

teyahb8, 2025-01-22:
Thank you for clarifying the issue with the test case. I disabled the optimization with `--opt-level=0` and I am no longer getting a segfault. However, can we consider the issue as a bug conditioned on the optimization level? I am asking because infinite loops might be used to continuously read from some resources and the default optimization level is (3) which is triggering the segfault. Your feedback is highly appreciated.

TianlongLiang, 2025-01-23:
> However, can we consider the issue as a bug conditioned on the optimization level?

The (some special kind, at least) infinite loop thing in the C/C++ world is more like undefined behavior:
https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2023/p2809r1.html#effect

I think your code compiles a similar native code from what the native code that C/C++ undefined behavior infinite loop will be compiled to, so in my opinion, it shouldn't be called a bug.

teyahb8, 2025-01-23:
I understand now. Thanks a lot for the clarification. I am closing the issue.

closed 2025-01-23 by teyahb8

TianlongLiang, 2025-02-06:
Sorry for my mistake, after some further investigation, I think this is a bug(fixed [here](https://github.com/bytecodealliance/wasm-micro-runtime/pull/4065)) and have a clearer view of the undefined behavior I mentioned above and want to share it with you:

**TL;DR**: no matter the wasm file or aot generated from the WAMR compiler(no matter the optimization level), WAMR runtime should run it in the same way that conforms to wasm spec. 

For the undefined behavior I mentioned before, consider this loop in C++:
```C++
int main() {
    bool done{false};
    while (!done);
    return 0;
}
```

The behavior will become well-defined when it's compiled to wasm, if it's O0, the infinite loop is kept while it becomes an unreachable opcode in O3. And aot compiler(in any optimization level) won't change wasm's behavior.

teyahb8, 2025-02-07:
I have checked. I am confirming the fix. Thanks for revisiting the bug and for explaining the undefined behavior.