wasm-micro-runtime/wasm-micro-runtime #4038: Compiling WASM to AOT loses the import table.

AY268, 2025-01-20:
I compiled a WASM file to an AOT file using WAMRC, which only has an import table. 

However, when I debug it, I find that `module->import_table_count` is always 0 in the file `wasm-micro-runtime/core/iwasm/aot/aot_loader.c`, the function `load_table_info`, while `module->table_count` is 1. 

This is how I compile it: `wamrc --bounds-checks=1 --target=aarch64 -o output.aot output.wasm`.

I wonder how this happens, and how I could solve it.

TianlongLiang, 2025-01-20:
The import table in aot isn't fully functional yet, which is why you see the discrepancy here, may I ask your use case? There may be a workaround for your use case

AY268, 2025-01-20:
I use this small WAT module:
`(module
  (import "env" "table" (table 10 funcref))
)`
and compile it to wasm:
`wat2wasm import_table.wat -o import_table.wasm`
then to aot:
`wamrc --bounds-checks=1 --target=aarch64 -o import_table.aot import_table.wasm`

AY268, 2025-01-20:
Also I meet another question when using AOT.
When I try to use a custom section, I find that it never enters the branch `load_custom_section` in the function `load_from_sections`(in file `wasm-micro-runtime/core/iwasm/aot/aot_loader.c`). 
I would like to know the reason for this. 

The module I used is:
`(module
  (custom "test" (data "\57\65\62\41\73\73\65\6D\62\6C\79"))
)`

TianlongLiang, 2025-01-20:
Hi, could you print more log for the custom section?
For example:
```bash
iwasm -v=5 test.aot
```

AY268, 2025-01-20:
Of course, I'd like to show you more logs. However, I compiled and ran it on Android, using libiwasm.so. Could you tell me how to print more logs for libiwasm.so?

More details about the custom section is that I found that it may not support compiling WAT with a custom section into WASM.
I get the wasm file from [customSections.any.js](https://github.com/WebAssembly/spec/blob/main/test/js-api/module/customSections.any.js), the `Custom sections with surrogate pairs` test. 
I only change the name of the custom section, and compile it to AOT myself.
I'm not sure whether this is the problem.

Really thank you for your help. If you have any ideas about the custom section issue or the import table issue, please let me know. Thanks again.

TianlongLiang, 2025-01-21:
About import table, it will be functional and used as your example after the development of [instnatiation linking](https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/instantiate_linking), which could take some time.

For printing logs, you can use `wasm_runtime_set_log_level(5)` to set log level to verbose

AY268, 2025-01-21:
Thanks for your answer. 
I can only use `wasm_runtime_set_log_level(WASM_LOG_LEVEL_VERBOSE);` to set the log level. 

For the custom section test, the log is as follows:
```
    I  [14:38:19:363 - 734F7D3ED0]: 
    I  Load module success.
    I  
    I  [14:38:19:363 - 734F7D3ED0]: 
    I  warning: wasm_runtime_free with NULL pointer
    I  
```

TianlongLiang, 2025-01-21:
About the custom section:
Did you add the option `--emit-custom-sections=` for wamrc? For details you can refer to `wamrc --help`:

```bash
  --emit-custom-sections=<section names>
                            Emit the specified custom sections to AoT file, using comma to separate
                            multiple names, e.g.
                                --emit-custom-sections=section1,section2,sectionN
```

I tested it on my local machine, and the aot file will have the custome section.

AY268, 2025-01-21:
This is actually the solution! Really thanks for your help!

AY268, 2025-01-22:
@TianlongLiang 
Hi, I still have a question about importing tables/memory in AOT. 
I found that when I compile the two WAT files to WASM, and then to AOT, they are different in the WASM files but the same in the AOT files.

```
(module
  (import "env" "memory" (memory 1 2))
)
```
```
(module
  (memory 1 2)
)
```
I don't know whether it's my fault that I should add some options but didn't, or if it is designed like this now. 

The command I used is as follows:
```
wat2wasm test.wat -o test.wasm
wamrc --bounds-checks=1 --target=aarch64 -o test.aot test.wasm
```
Could you give some help again?

TianlongLiang, 2025-01-23:
Basically, if you want to import and share the globals/tables/memories among your wasm apps, it needs [instnatiation linking](https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/instantiate_linking), and it isn't supported now. 

And if you don't have share demand, treating an import as a non-import is the same thing, that's what WAMR does now(in aot compiler) and it complies with your observation.