wasm-micro-runtime/wasm-micro-runtime #4041: Command call_indirect can only call index 0 in AOT

AY268, 2025-01-21:
I found that if I use the `call_indirect` command in AOT, it always runs well when calling the first function in the table. However, when I call the second function in the table, it crashes.

This is the test case I use:
```
(module
  (func $add_one (param $x i32) (result i32)
    local.get $x
    i32.const 1
    i32.add
  )

  (table 2 funcref)
  (elem (i32.const 0) $add_one $add_one)

  (func (export "call_first_func") (param $x i32) (result i32)
    local.get $x
    i32.const 0
    call_indirect (type $add_one)
  )

  (func (export "call_second_func") (param $x i32) (result i32)
    local.get $x
    i32.const 1
    call_indirect (type $add_one)
  )

  (type $add_one (func (param i32) (result i32)))
)
```

And this is how I compile it to AOT:
```
wat2wasm test.wat -o test.wasm
wamrc --bounds-checks=1 --target=aarch64 -o test.aot test.wasm
```

I build WAMR to run on Android, with `WAMR_DISABLE_HW_BOUND_CHECK=1` and `WAMR_BUILD_LOAD_CUSTOM_SECTION=1`.

TianlongLiang, 2025-01-21:
It could be platform specific issue, I can't reproduce it on my x86 Ubuntu. BTW, how do you call the function? IIUC, you are embedding WAMR and using wasm _call_function to call the function instead of `iwasm -f call_second_func test.aot 5`, right?

AY268, 2025-01-21:
Sure, I use `wasm _call_function` to call the function.
The whole process is that I use `wasm_module_new` and `wasm_instance_new_with_args` to instantiate an AOT module. Then I use the `wasm_instance_exports` function to get the exported function. Finally, I use `wasm _call_function` to call the function.

The stack trace when the crash occurs is as follows:
[image]

And it occurs in file `wasm-micro-runtime/core/iwasm/aot/aot_runtime.c`, function invoke_native_internal:
[image]

AY268, 2025-01-22:
I'm sorry to say that this fault is caused by my changes.

closed 2025-01-22 by AY268