wasm-micro-runtime/wasm-micro-runtime #4045: 关于 Alloc_With_Pool 模式与 linear memory 的问题

lxbxl, 2025-01-22:
在 [《The WAMR memory model》](https://bytecodealliance.github.io/wamr.dev/blog/the-wamr-memory-model/#memory-allocators) 中提到 Alloc_With_Pool 模式会将 wamr 使用的所有内存都交由 ems 从指定内存块（RuntimeInitArgs.mem_alloc_option.pool.heap_buf）中分配。

而在 [《Build WAMR vmcore》](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#user-defined-linear-memory-allocator) 中提到 linear memory 默认由系统分配。

从源码来看 Alloc_With_Pool 模式最终使用 os_mmap 来分配 linear memory。**这是否意味着  《The WAMR memory model》 描述有误？还是说 wasm_allocate_linear_memory 函数需适配一下 Alloc_With_Pool 模式？**

我正尝试在 windows 内核环境下使用 wamr 。由于在 windows 的中断响应函数中不能进行内存分配的操作，这导致我必须在一开始就要分配好整个 wamr 所使用的内存。请问这种场景的最佳实践是什么？（若 Alloc_With_Pool 模式如 《The WAMR memory model》中说的那样，就非常符合这个场景。）我能想到的解决方案是使用 Alloc_With_Allocator 模式，并自己实现一个内存池用于 Alloc_With_Allocator 的 MemAllocOption，并且对于诸如 os_malloc os_mmap 之类内存相关的 os_xxx 函数也需用这个内存池实现。

TianlongLiang, 2025-01-22:
> Alloc_With_Pool 模式最终使用 os_mmap 来分配 linear memory

对，Alloc_With_Pool 的时候还是用os_mmap, 只不过一些小平台上比如zephyr，os_mmap 会定义成 BH_MALLOC，这个时候内存都会从 global_heap_buf里来。

我感觉 wasm_allocate_linear_memory 函数适配一下 Alloc_With_Pool 模式就行了，Alloc_With_Pool 的时候（并且得同时 -DWAMR_DISABLE_HW_BOUND_CHECK=1 因为没在用os_mmap去提供 hardware boundary check了) 调用 wasm_runtime_malloc 而不是wasm_mmap_linear_memory, 这样应该就能满足你的需求了

lxbxl, 2025-01-23:
> 对，Alloc_With_Pool 的时候还是用os_mmap, 只不过一些小平台上比如zephyr，os_mmap 会定义成 BH_MALLOC，这个时候内存都会从 global_heap_buf里来。
>
> 我感觉 wasm_allocate_linear_memory 函数适配一下 Alloc_With_Pool 模式就行了，Alloc_With_Pool 的时候（并且得同时 [...]

嗯嗯，我的一个临时方案就是让 os_mmap 直接调 wasm_runtime_malloc。像 zephyr 这些小平台，用 BH_MALLOC 会更优雅些。

lum1n0us, 2025-01-24:
> 由于在 windows 的中断响应函数中不能进行内存分配的操作，这导致我必须在一开始就要分配好整个 wamr 所使用的内存

可以考虑通过提前加载 wasm 文件并实例化的方式，将“中断响应函数” 对应的 “wasm加载，实例化和执行wasm函数” 变成 “执行wasm函数”。这样可以有效减少需要管理的 “wamr使用的内存”，变成“wasm函数使用的内存”.

closed 2025-02-10 by lxbxl