wasm-micro-runtime/wasm-micro-runtime #4066: Compile Wamr product mini on raspberry pi 4

Papadiddypub, 2025-02-06:
Hi,
I downloaded WAMR and wanted to compile it on the raspberry pi 4. When compiling i get the attached error ([error.log](https://github.com/user-attachments/files/18692575/error.log) 

I build form the following location using cmake: 
~/wamr/product-mini/platforms/linux/build $ : cmake ..
~/wamr/product-mini/platforms/linux/build $ : make

environment (64bit pi 4 with gcc 12.2):
- Linux WASMraspberrypi 6.6.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux
- gcc (Raspbian 12.2.0-14+rpi1) 12.2.0
- Package: build-essential, Version: 12.9

Information:
I set up a an environment on the raspberry pi 3 which compiles with the same environment (also AARCH64) without a problem. 
I looked at Issue #1993 and running: 
cmake -DWAMR_BUILD_SIMD=0 -DWAMR_BUILD_TARGET=AARCH64V8 ..  
- I get an error in CMakeFiles/vmlib.dir/home/pi/wamr/core/iwasm/common/arch/invokeNative_aarch64.s.o

TianlongLiang, 2025-02-07:
So what kind of error `cmake -DWAMR_BUILD_SIMD=0 -DWAMR_BUILD_TARGET=AARCH64V8` gives?

TianlongLiang, 2025-02-07:
You can try add `add_compile_options(-march=armv8-a+simd)` in CMakeLists.txt to see whether it helps with SIMD asm file compilation error.

Papadiddypub, 2025-02-07:
@TianlongLiang thanks for your support. 
I foudn the challenge in the toolchain. A different gcc compiler was linked to the gcc symlink and which wasn't compiling to AARCH64. 
The raspberry pi compiler aarch64-linux-gnu-gcc works. 

Ticket can be closed

closed 2025-02-07 by Papadiddypub