wasm-micro-runtime/wasm-micro-runtime #4081: wasm_runtime_load 未初始化 WASIArguments 的 stdio 内容

lxbxl, 2025-02-13:
无论是 wasm_load 还是 aot_load_from_aot_file，都未调用 os_get_invalid_handle 来初始化 stdio 数组项。这导致 Windows 平台下 wasm_runtime_instantiate 内调用 wasm_runtime_init_wasi 时 stdinfd stdoutfd stderrfd 传入了 0。[win_file.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/windows/win_file.c) 中使用的 INVALID_HANDLE_VALUE 值为 -1。

lum1n0us, 2025-02-18:
Thanks for reporting it. #4092 is supposed to fix that. Please review it.

lxbxl, 2025-02-18:
> Thanks for reporting it. [#4092](https://github.com/bytecodealliance/wasm-micro-runtime/pull/4092) is supposed to fix that. Please review it.

os_get_invalid_handle 的返回类型不是原始的文件句柄。 https://github.com/bytecodealliance/wasm-micro-runtime/pull/4092#issuecomment-2664634319

PR #4092 (2025-05-08): initialize WASI stdio handles to invalid for better error handling
fix #4081
closed 2025-02-19 by lxbxl