wasm-micro-runtime/wasm-micro-runtime #4084: WAMR behaves differently compared to other runtimes for a particular test case.

teyahb8, 2025-02-14:
### Subject of the issue

I have generated a test case and tested multiple runtimes with it. WAMR (AOT) mode behaves differently compared to other runtimes for this particular test case. Other runtimes including WAMR (Interpreter) produce `import error` whereas WAMR (AOT) produces a segfault.

### Test case

The .wat (.txt) file is as follows: [21.txt](https://github.com/user-attachments/files/18792656/21.txt). I could not upload this file as a .wat file because of file extension restrictions.

### Your environment

* WAMR version (2.2.0), platform (Linux), cpu architecture (X86_64), running mode (AOT).

### Steps to reproduce

Rename `21.txt` to `21.wat`
Run `wat2wasm 21.wat` to generate `21.wasm`
Run `wamrc --bounds-checks=1 -o 21.aot 21.wasm` to generate `21.aot`
Run `iwasm -f main 21.aot` to manifest the crash

### Expected behavior

Other runtimes like wasmtime, wasmer, wamr (interpreter), and wasmedge all produce `import error` with this test case.

### Actual behavior

Outputs from `wamrc` and `iwasm` are as follows:

wamrc output:

```
Create AoT compiler with:
  target:        x86_64
  target cpu:    znver3
  target triple: x86_64-unknown-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file 21.aot was generated.
```

iwasm output:

```
warning: failed to link import function (, )
unhandled SIGSEGV, si_addr: (nil)
```

Can you confirm if this is a bug? Thanks in advance.

lum1n0us, 2025-02-18:
The difference in output between iwasm and wamrc is a deliberate design decision. Rather than failing during instantiation, WAMR opts to attempt executing Wasm functions until it encounters an unlinked import function. This is why, when the import requirements are not fully met:
- There is no error reported during initialization.
- AOT files are still generated.

BTW, The SIGSEGV is suspicious. Could you please tell me which version of iwasm you are using?

teyahb8, 2025-02-18:
I see. Thanks for explaining the design decision. But, this test case might be a corner case to cause a segfault instead of gracefully exiting. I am building `iwasm` from the latest commit. I am sharing the commit and version information in the attached screenshot.

[image]

teyahb8, 2025-07-11:
I will appreciate any update on this crash. Many thanks.