wasm-micro-runtime/wasm-micro-runtime #4088: build wasm file to AOT, report error Unsupported opcode: 0x8 WASM_OP_THROW

ericCHensssssssss, 2025-02-15:
when i build wasm file to aot, it report errors. it seems not support WASM_OP_THROW

 ./wamrc --enable-gc -o test.aot --target=armv7 test.wasm

Create AoT compiler with:
  target:        armv7
  target cpu:
  target triple: armv7-pc-linux-gnu
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file
Unsupported opcode: 0x8

yamt, 2025-02-17:
it's expected because currently exception-handling is only implemented for the classic interpreter.

ericCHensssssssss, 2025-02-17:
Thanks for your quick response，i compile Kotlin to WebAssembly ，do you plan to support the exception-handling  feature in AOT，

lum1n0us, 2025-02-17:
I haven't heard of any further plans from the community regarding exception handling in AOT. This is partly because the exception-handling proposal underwent significant changes at the end of last year, and the version implemented by WAMR is now considered "legacy exception handling." This makes it difficult to extend the current interpreter's solution to AOT mode.

ericCHensssssssss, 2025-02-17:
i want run Kotlin program  at embbed devices，so aot is aot is really  important for me，if not support aot，it will slow use classic interpreter

lum1n0us, 2025-02-17:
Is there a way, like some command line options, for the Kotlin toolchain to compile .kt files to .wasm without utilizing exception handling opcodes?

ericCHensssssssss, 2025-02-17:
ok. goog idea, it can change the kotlin compiler chain . 
If i remove exception handling . is AOT support GC feature?

ericCHensssssssss, 2025-02-17:
@lum1n0us   when i build wasm from Kotlin，use the follow option              "-Xwasm-use-traps-instead-of-exceptions",
    wasmrc can compile the wasm to aot file success。
    how to run the aot，  
     when i run     ./iwasm  Sys.aot   
i meet the error：
 warning: failed to allocate memory for gc object
AOT module instantiate failed: create struct object failed

ericCHensssssssss, 2025-02-17:
@lum1n0us  Thanks. it worked when i removed the EH,  

and how can i run the aot. is the same as wasm file 

./iwasm -f test Sys.aot 12

when i run it. error produced

unhandled SIGSEGV, si_addr: 0x53
zsh: abort      ./iwasm -f test  Sys.aot 12

ericCHensssssssss, 2025-02-19:
when i run the aot file use the follow command 
./iwasm Sys.aot
report the error as the screen shot i debug.
bh_assert(rtt_type->type_flag == WASM_TYPE_STRUCT);

[image]

lum1n0us, 2025-02-21:
@ericCHensssssssss would you mind uploading `sys.wasm` and/or `sys.aot`?

chenliuyang1989, 2025-02-22:
The type_idx is fixed after[https://github.com/bytecodealliance/wasm-micro-runtime/pull/4094]

but meet new error
 ./iwasm -f test aqaraSys.aot
Exception: null function reference

[压缩.zip](https://github.com/user-attachments/files/18918014/default.zip)
[aotfile.zip](https://github.com/user-attachments/files/18918015/aotfile.zip)

ericCHensssssssss, 2025-02-24:
Any finds？

lum1n0us, 2025-02-24:
#4094 from @chenliuyang1989 is supposed to fix the issue. Please give it a try.

lum1n0us, 2025-02-24:
I'm using the latest commit d9c01b39d11a068558363b20be39c7364d871322 from the main branch.

I compiled iwasm with the flag `-DWAMR_BUILD_GC=1`.
When running it with a larger garbage collection heap size, like `--gc-heap-size=1073741824`, I received endless output. Is this the expected behavior?

``` bash
$ ./build/iwasm --gc-heap-size=1073741824   -f test   ./aqaraSys-wasm-wasi.wasm

...
this is debug message: before propagate:out - AStatusNumeric,value:0.11333333333333334 {Status.ok}
没有link，links大小:1.开始传递至父节点，p:out,v:0.11333333333333334 {Status.ok}
this is debug message: before propagate:_value - ADouble,value:0.11333333333333334
没有link，links大小:0.开始传递至父节点，p:_value,v:0.11333333333333334
working值变更了,name:Ramp$288$29,working:0.11333333333333334 {Status.ok},out:0.10666666666666667 {Status.ok}
this is debug message: before propagate:_value - ADouble,value:0.10666666666666667
没有link，links大小:0.开始传递至父节点，p:_value,v:0.10666666666666667
this is debug message: before propagate:out - AStatusNumeric,value:0.11333333333333334 {Status.ok}
没有link，links大小:1.开始传递至父节点，p:out,v:0.11333333333333334 {Status.ok}
this is debug message: before propagate:_value - ADouble,value:0.11333333333333334
没有link，links大小:0.开始传递至父节点，p:_value,v:0.11333333333333334
working值变更了,name:Ramp$289$29,working:0.11333333333333334 {Status.ok},out:0.10666666666666667 {Status.ok}
this is debug message: before propagate:_value - ADouble,value:0.10666666666666667
没有link，links大小:0.开始传递至父节点，p:_value,v:0.10666666666666667
this is debug message: before propagate:out - AStatusNumeric,value:0.11333333333333334 {Status.ok}
没有link，links大小:1.开始传递至父节点，p:out,v:0.11333333333333334 {Status.ok}
this is debug message: before propagate:_value - ADouble,value:0.11333333333333334
没有link，links大小:0.开始传递至父节点，p:_value,v:0.11333333333333334
working值变更了,name:Ramp$2810$29,working:0.11333333333333334 {Status.ok},out:0.10666666666666667 {Status.ok}
this is debug message: before propagate:_value - ADouble,value:0.10666666666666667
没有link，links大小:0.开始传递至父节点，p:_value,v:0.10666666666666667
this is debug message: before propagate:out - AStatusNumeric,value:0.11333333333333334 {Status.ok}
没有link，links大小:1.开始传递至父节点，p:out,v:0.11333333333333334 {Status.ok}

```

chenliuyang1989, 2025-02-24:
yes， this is expected behavior. wasm worked as expect

chenliuyang1989, 2025-02-24:
do you produce the null function reference error when run aot file?

lum1n0us, 2025-02-26:
> do you produce the null function reference error when run aot file?

No. I can't reproduce the issue with the commit https://github.com/bytecodealliance/wasm-micro-runtime/commit/d9c01b39d11a068558363b20be39c7364d871322 from the main branch.

chenliuyang1989, 2025-02-26:
you can run the aot file with no exception？

This is the my os
Linux VM-16-15-ubuntu 5.15.0-119-generic #129-Ubuntu SMP Fri Aug 2 19:25:20 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

i  meet the follow error.  
#00: 0x8e906 - $f1616
#01: 0x10c90d - myTest

Exception: null function reference test

what more logs need i provide please let me know,Thanks

chenliuyang1989, 2025-03-03:
@lum1n0us  can you produce new error when run the aot？  or do you need some more logs