wasm-micro-runtime/wasm-micro-runtime #4097: For  AOT,If enable gc, the reference_type will disable,Why ?

chenliuyang1989, 2025-02-21:
[image]

TianlongLiang, 2025-02-21:
The AOT option here: ref types here refer to wasm [ref types proposal](https://github.com/WebAssembly/reference-types/blob/master/proposals/reference-types/Overview.md) which became core wasm spec, gc refer to [gc proposal](https://github.com/WebAssembly/gc/blob/main/proposals/gc/Overview.md). As those two proposals specified, some opcodes (for example,  call_indirect) have different meanings and need different implementations, therefore the flag control here

chenliuyang1989, 2025-02-21:
Thank you for your explanation. I'm in a hurry to let aot work. But unfortunately there have been problems。

If i enable [gc proposal](https://github.com/WebAssembly/gc/blob/main/proposals/gc/Overview.md). so [ref types](https://github.com/WebAssembly/reference-types/blob/master/proposals/reference-types/Overview.md) is diabled。

my wasm file is build from Kotlin，but it has ref types。 if the refTypes is disabled，how can my aot work well？

chenliuyang1989, 2025-02-21:
Here's how I do it

First：build aot

./wamrc --enable-gc -o --bounds-checks=1 aqaraSys.aot aqaraSys-wasm-wasi.wasm

Second：run aot
./iwasm -f test aqaraSys.aot

third：meet error
./iwasm -f test aqaraSys.aot
Exception: null function reference

**if i run wasm with classic interpret，it work well**

TianlongLiang, 2025-02-21:
It's not disabled but redefine the ref types to conform to the GC proposal definition, there shouldn't be any error. Could you please attach the WASM file?

chenliuyang1989, 2025-02-21:
ok.please help check the file ,thanks . the entry function is test

[压缩.zip](https://github.com/user-attachments/files/18903727/default.zip)

chenliuyang1989, 2025-02-21:
add aot file

[aotfile.zip](https://github.com/user-attachments/files/18903756/aotfile.zip)

ericCHensssssssss, 2025-02-23:
I write one simple test，If i use println  method in Kotlin ， it will produce  null function reference。

The one use println ：will produce  null function reference。

[image]

remove println ，no error

[image]

chenliuyang1989, 2025-02-24:
@TianlongLiang  any updates？ Very eager to get it up and running。thanks!!

TianlongLiang, 2025-02-26:
No luck yet,  I was busy with some other things the last few days. I tried your wasm file in LLVM JIT running mode which is fine, and in aot the latest commit will core dump, I think maybe a bug in the aot emitting type info. I will try the simple test to see where was wrong. Keep you posted

chenliuyang1989, 2025-02-26:
Thanks,

I'm stuck here now, and I need to get to a conclusion as soon as possible

when i update the latest main branch codes。 i find the same segment error, but did not find the root case  .a a a

chenliuyang1989, 2025-02-28:
@TianlongLiang  would you help have a look this week？  i really need it

chenliuyang1989, 2025-02-28:
[

[simplestWasm&wat.zip](https://github.com/user-attachments/files/19023157/simplestWasm.wat.zip)

]

This is the simple and small wasm file to product the error。
There are two export functions myTest（segment error） and myTest2

[image]

lum1n0us, 2025-03-12:
This has prompted a feature request for enhanced support of struct fields in AOT mode, which may take a considerable amount of time to implement.

chenliuyang1989, 2025-03-12:
Thanks~！Excuse me, when can it be finished? An approximate time is ok