wasm-micro-runtime/wasm-micro-runtime #4101: Windows wamrc.exe Segmentation fault

calvin2021y, 2025-02-23:
I download  https://github.com/bytecodealliance/wasm-micro-runtime/releases/download/WAMR-2.2.0/wamrc-2.2.0-x86_64-windows-latest.tar.gz and run:

```sh
./wamrc.exe --format=aot -o fib.aot fib.wasm
Segmentation fault
```

I try `fib.aot` generate from darwin x64 wamrc, and run on linux x64.  it work.

Can I use this fib.aot to windows x64?   

current I can not make windows version to work, the app crashed:
```sh
 thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ff8f57f46f8: Access violation writing location 0x2c2ceae0004
    frame #0: 0x00007ff8f57f46f8 msvcrt.dll`memset + 120
msvcrt.dll`memset:
->  0x7ff8f57f46f8 <+120>: movups %xmm0, (%rcx)
    0x7ff8f57f46fb <+123>: addq   %rcx, %r8
    0x7ff8f57f46fe <+126>: addq   $0x10, %rcx
    0x7ff8f57f4702 <+130>: andq   $-0x10, %rcx
```

calvin2021y, 2025-02-23:
```sh
./iwasm.exe                  
Segmentation fault
```

TianlongLiang, 2025-02-26:
I list all the dynamic library needed for iwasm and wamrc:
```bash
File Type: EXECUTABLE IMAGE

  Image has the following dependencies:

    ntdll.dll
    ADVAPI32.dll
    KERNEL32.dll
    api-ms-win-core-path-l1-1-0.dll
    WS2_32.dll
    bcrypt.dll
    MSVCP140.dll
    VCRUNTIME140.dll
    VCRUNTIME140_1.dll
    api-ms-win-crt-heap-l1-1-0.dll
    api-ms-win-crt-convert-l1-1-0.dll
    api-ms-win-crt-string-l1-1-0.dll
    api-ms-win-crt-stdio-l1-1-0.dll
    api-ms-win-crt-filesystem-l1-1-0.dll
    api-ms-win-crt-runtime-l1-1-0.dll
    api-ms-win-crt-math-l1-1-0.dll
    api-ms-win-crt-utility-l1-1-0.dll
    api-ms-win-crt-time-l1-1-0.dll
    api-ms-win-crt-environment-l1-1-0.dll
    api-ms-win-crt-locale-l1-1-0.dll

  Image has the following delay load dependencies:

    SHELL32.dll
    ole32.dll
```

```bash

File Type: EXECUTABLE IMAGE

  Image has the following dependencies:

    ADVAPI32.dll
    KERNEL32.dll
    MSVCP140.dll
    api-ms-win-core-path-l1-1-0.dll
    WS2_32.dll
    bcrypt.dll
    VCRUNTIME140.dll
    VCRUNTIME140_1.dll
    api-ms-win-crt-heap-l1-1-0.dll
    api-ms-win-crt-convert-l1-1-0.dll
    api-ms-win-crt-string-l1-1-0.dll
    api-ms-win-crt-stdio-l1-1-0.dll
    api-ms-win-crt-environment-l1-1-0.dll
    api-ms-win-crt-runtime-l1-1-0.dll
    api-ms-win-crt-filesystem-l1-1-0.dll
    api-ms-win-crt-math-l1-1-0.dll
    api-ms-win-crt-utility-l1-1-0.dll
    api-ms-win-crt-time-l1-1-0.dll
    api-ms-win-crt-locale-l1-1-0.dll

  Image has the following delay load dependencies:

    SHELL32.dll
    ole32.dll
```

I think after you install vscode studio toolchain it should be fine

calvin2021y, 2025-02-26:
after install https://aka.ms/vs/17/release/vc_redist.x64.exe,  it work

closed 2025-02-26 by calvin2021y