wasm-micro-runtime/wasm-micro-runtime #4103: add x86_64-w64-windows-gnu aot triple support

calvin2021y, 2025-02-24:
current wamrc only support mscv  triple,   can not made it work with x86_64-w64-windows-gnu.  (you can get this from https://github.com/mstorsjo/llvm-mingw/releases/download/20250114/llvm-mingw-20250114-msvcrt-ubuntu-20.04-x86_64.tar.xz)

calvin2021y, 2025-02-24:
I guess allow pass triple from arguments for wamrc will be a lot easy, and support all future arch & target.

if pass triple, then arch and target can be infered.

TianlongLiang, 2025-02-26:
You can use option `--target-abit=gnu` to change target abi, should it be sufficient for your usage?
```bsh
./wamrc.exe --target-abi=help
Supported ABI: gnu eabi eabihf gnueabihf msvc ilp32 ilp32f ilp32d lp64 lp64f lp64d
```

calvin2021y, 2025-02-26:
I guess it must be `x86_64-w64-windows-gnu`,  with `--target-abit=gnu` I get `x86_64-pc-linux-gnu`

 ```sh
./wamrc.exe --target-abi=gnu --target=x86_64 --enable-tail-call --format=aot -o fib.win fib.wasm
Create AoT compiler with:           
  target:        x86_64             
  target cpu:                       
  target triple: x86_64-pc-linux-gnu
  cpu features:                     
  opt level:     3                  
  size level:    3                  
  output format: AoT file           
Compile success, file fib.win was generated.
```

crashed when  load aot

```sh
./tests_wamr.exe             
=================================================================
==3320==ERROR: AddressSanitizer: access-violation on unknown address 0x01cf501c0004 (pc 0x7ff8f68f4678 bp 0x006411efb8b0 sp 0x006411efb828 T0)
==3320==The signal is caused by a WRITE memory access.                                                                                        
    #0 0x7ff8f68f4677 in memset+0x77 (C:\Windows\SYSTEM32\ntdll.dll+0x1800a4677)                                                              
    #1 0x7ff8b3ff2cc0 in _asan_memset+0x90 (C:\Users\admin\libclang_rt.asan_dynamic-x86_64.dll+0x180042cc0)                                   
    #2 0x7ff7f0b34206  (C:\Users\admin\tests_wamr.exe+0x140214206)                                                                            
    #3 0x7ff7f0b34c67  (C:\Users\admin\tests_wamr.exe+0x140214c67)                                                                            
    #4 0x7ff7f0ac02ec  (C:\Users\admin\tests_wamr.exe+0x1401a02ec)                                                                            
    #5 0x7ff7f0a5f2d5  (C:\Users\admin\tests_wamr.exe+0x14013f2d5)                                                                            
    #6 0x7ff7f0a5303a  (C:\Users\admin\tests_wamr.exe+0x14013303a)
    #7 0x7ff7f0a5033a  (C:\Users\admin\tests_wamr.exe+0x14013033a)
    #8 0x7ff7f0999403  (C:\Users\admin\tests_wamr.exe+0x140079403)
    #9 0x7ff7f0999493  (C:\Users\admin\tests_wamr.exe+0x140079493)
    #10 0x7ff7f097a5e7  (C:\Users\admin\tests_wamr.exe+0x14005a5e7)
    #11 0x7ff7f092133b  (C:\Users\admin\tests_wamr.exe+0x14000133b)
    #12 0x7ff7f0921395  (C:\Users\admin\tests_wamr.exe+0x140001395)
    #13 0x7ff8f5597373 in BaseThreadInitThunk+0x13 (C:\Windows\System32\KERNEL32.DLL+0x180017373)
    #14 0x7ff8f689cc90 in RtlUserThreadStart+0x20 (C:\Windows\SYSTEM32\ntdll.dll+0x18004cc90)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: access-violation (C:\Windows\SYSTEM32\ntdll.dll+0x1800a4677) in memset+0x77
==3320==ABORTING

```

TianlongLiang, 2025-02-26:
This is due to `vendor_sys` using the one defined in `LLVMGetDefaultTargetTriple`, it seems that the default LLVM build for `vendor_sys` is Linux. Is building wamrc with the llvm library link you provided an option for you? If so, you can try whether it solves your problem

calvin2021y, 2025-02-26:
The wamrc.exec default is `x86_64-pc-windows-msvc` ( without pass target-abi and target),  use `x86_64-pc-windows-msvc` crash for me too.

```sh
./wamrc.exe --target-abi=msvc --enable-tail-call --format=aot -o fib.win fib.wasm
Create AoT compiler with:
  target:        x86_64
  target cpu:
  target triple: x86_64-pc-windows-msvc
  cpu features:
  opt level:     3
  size level:    1
  output format: AoT file
Compile success, file fib.win was generated.
```

I guess problem is from here,   allow user set vendor_sys from arguments will be much simple and prevent more questions(like require android/ios/musl triple).

```c
            /**
             * Set <vendor>-<sys> according to abi to generate the object file
             * with the correct file format which might be different from the
             * default object file format of the host, e.g., generating AOT file
             * for Windows/MacOS under Linux host, or generating AOT file for
             * Linux/MacOS under Windows host.
             */

            if (!strcmp(abi, "msvc")) {
                if (!strcmp(arch1, "i386"))
                    vendor_sys = "-pc-win32-";
                else
                    vendor_sys = "-pc-windows-";
            }
            else {
                if (is_baremetal_target(arch, cpu, abi))
                    vendor_sys = "-unknown-none-";
                else
                    vendor_sys = "-pc-linux-";
            }
```

TianlongLiang, 2025-02-26:
You are right, the correct way is to add the command line option to control vendor_sys(add some extra check to make sure triple is compatible too)

calvin2021y, 2025-02-26:
I am not sure this related: 

```sh
(lldb) bt
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ff8f57f46f8: Access violation writing location 0x15b201a0004
  * frame #0: 0x00007ff8f57f46f8 msvcrt.dll`memset + 120
    frame #1: 0x00007ff73ab78404 tests_wamr.exe`mem_allocator_create_with_struct_and_pool
    frame #2: 0x00007ff73ab6ed73 tests_wamr.exe`aot_instantiate
    frame #3: 0x00007ff73ab4f0c3 tests_wamr.exe`main
    frame #4: 0x00007ff73ab3133c tests_wamr.exe`__tmainCRTStartup at crtexe.c:266:15
    frame #5: 0x00007ff73ab31396 tests_wamr.exe`mainCRTStartup at crtexe.c:186:9
    frame #6: 0x00007ff8f5597374 kernel32.dll`BaseThreadInitThunk + 20
    frame #7: 0x00007ff8f689cc91 ntdll.dll`RtlUserThreadStart + 33
```

calvin2021y, 2025-02-26:
with san:

```sh
(lldb) bt
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ff8f68f4678: Access violation writing location 0x127c53930004
  * frame #0: 0x00007ff8f68f4678 ntdll.dll`memset + 120
    frame #1: 0x00007ff8b0242cc1 libclang_rt.asan_dynamic-x86_64.dll`::__asan_memset() at asan_interceptors_memintrinsics.cpp:67:3
    frame #2: 0x00007ff673e54207 tests_wamr.exe`gc_init_internal
    frame #3: 0x00007ff673e54c68 tests_wamr.exe`gc_init_with_struct_and_pool
    frame #4: 0x00007ff673de02ed tests_wamr.exe`mem_allocator_create_with_struct_and_pool
    frame #5: 0x00007ff673d7f2d6 tests_wamr.exe`memory_instantiate
    frame #6: 0x00007ff673d7303b tests_wamr.exe`memories_instantiate
    frame #7: 0x00007ff673d7033b tests_wamr.exe`aot_instantiate
    frame #8: 0x00007ff673cb9404 tests_wamr.exe`wasm_runtime_instantiate_internal
    frame #9: 0x00007ff673cb9494 tests_wamr.exe`wasm_runtime_instantiate
    frame #10: 0x00007ff673c9a5e8 tests_wamr.exe`main
    frame #11: 0x00007ff673c4133c tests_wamr.exe`__tmainCRTStartup at crtexe.c:266:15
    frame #12: 0x00007ff673c41396 tests_wamr.exe`mainCRTStartup at crtexe.c:186:9
    frame #13: 0x00007ff8f5597374 kernel32.dll`BaseThreadInitThunk + 20
    frame #14: 0x00007ff8f689cc91 ntdll.dll`RtlUserThreadStart + 33
```

gc is disabled

calvin2021y, 2025-02-26:
```sh
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x1647bf83edc: User-mode data execution prevention (DEP) violation at location 0x1647bf83edc
  * frame #0: 0x000001647bf83edc
    frame #1: 0x00007ff7de9a64ea tests_wamr.exe`invoke_ii_i
    frame #2: 0x00007ff7de9ede12 tests_wamr.exe`invoke_native_internal
    frame #3: 0x00007ff7de9ed762 tests_wamr.exe`aot_call_function
    frame #4: 0x00007ff7de92d495 tests_wamr.exe`wasm_runtime_call_wasm
    frame #5: 0x00007ff7de92f77b tests_wamr.exe`wasm_runtime_call_wasm_a
    frame #6: 0x00007ff7de90aab2 tests_wamr.exe`main
    frame #7: 0x00007ff7de8b133c tests_wamr.exe`__tmainCRTStartup at crtexe.c:266:15
    frame #8: 0x00007ff7de8b1396 tests_wamr.exe`mainCRTStartup at crtexe.c:186:9
    frame #9: 0x00007ff8f5597374 kernel32.dll`BaseThreadInitThunk + 20
    frame #10: 0x00007ff8f689cc91 ntdll.dll`RtlUserThreadStart + 33
```

calvin2021y, 2025-03-01:
and armv7a-unknown-linux-android will not work for any target,  report expect arm32, but get armv7.  also try thumbv7 not work

lum1n0us, 2025-04-14:
My proposal https://github.com/bytecodealliance/wasm-micro-runtime/pull/4199

lum1n0us, 2025-05-06:
@calvin2021y #4199 has been merged. Please give it a try and let me know your comments.

calvin2021y, 2025-05-06:
thanks for the great work.

use the main branch with https://github.com/bytecodealliance/wasm-micro-runtime/pull/4213,  simple example with aot work great for all platform.

When I test the `wasm-wasi` target,  the `iwasm` get error for wasm file:

```sh
WASM module load failed: unsupported opcode 12
```

for aot format:

```sh
wamrc --disable-simd --enable-tail-call --format=aot --target=x86_64-pc-linux-gnu --enable-segue --size-level=0 -o wasi_test.aot wasi_test.wasm
```

```sh
Register native, size: 32
Register native, size: 32
Register native, size: 32
Register native, size: 32
[MEM] wasm_engine_new_internal(...) L393
===== memory usage =====
VmPeak:	21474890040 kB
VmSize:	21474890040 kB
VmLck:	       0 kB
VmPin:	       0 kB
VmHWM:	   13156 kB
VmRSS:	   13156 kB
RssAnon:	    9860 kB
RssFile:	    3296 kB
RssShmem:	       0 kB
VmData:	15032431580 kB
VmStk:	     136 kB
VmExe:	    2228 kB
VmLib:	    2080 kB
VmPTE:	     636 kB
VmSwap:	       0 kB
==========
[MEM] wasm_engine_new_internal(...) L428
===== memory usage =====
VmPeak:	21474890040 kB
VmSize:	21474890040 kB
VmLck:	       0 kB
VmPin:	       0 kB
VmHWM:	   13156 kB
VmRSS:	   13156 kB
RssAnon:	    9860 kB
RssFile:	    3296 kB
RssShmem:	       0 kB
VmData:	15032432092 kB
VmStk:	     136 kB
VmExe:	    2228 kB
VmLib:	    2080 kB
VmPTE:	     672 kB
VmSwap:	       0 kB
==========
[MEM] wasm_store_new(...) L664
===== memory usage =====
VmPeak:	21474890040 kB
VmSize:	21474890040 kB
VmLck:	       0 kB
VmPin:	       0 kB
VmHWM:	   13496 kB
VmRSS:	   13496 kB
RssAnon:	   10072 kB
RssFile:	    3424 kB
RssShmem:	       0 kB
VmData:	15032432092 kB
VmStk:	     136 kB
[... 130 lines cut ...]
VmSwap:	       0 kB
==========
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1807243==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000500 (pc 0x000041cb6116 bp 0x7ffd8585cac0 sp 0x7ffd8585ca70 T0)
==1807243==The signal is caused by a READ memory access.
==1807243==Hint: address points to the zero page.
    #0 0x000041cb6116  (<unknown module>)
    #1 0x55fc608995e8 in wasm_runtime_invoke_native wamr/core/iwasm/common/wasm_runtime_common.c:6069:9
    #2 0x55fc6092c424 in invoke_native_internal wamr/core/iwasm/aot/aot_runtime.c:2410:12
    #3 0x55fc6092be9e in aot_call_function wamr/core/iwasm/aot/aot_runtime.c:2630:15
    #4 0x55fc6088d89f in wasm_runtime_call_wasm wamr/core/iwasm/common/wasm_runtime_common.c:2606:15
    #5 0x55fc60862e31 in wasm_func_call wamr/core/iwasm/common/wasm_c_api.c:3434:10
    #6 0x55fc60821b6a in main /opt/test.c:278:9
    #7 0x7f0c89aab249 in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16

==1807243==Register values:
rax = 0x000055fc608a3aa8  rbx = 0x00007c7c88ee0080  rcx = 0x0000000000000000  rdx = 0x0000000000000000  
rdi = 0x00007b0c86a7d800  rsi = 0x0000000000000000  rbp = 0x00007ffd8585cac0  rsp = 0x00007ffd8585ca70  
 r8 = 0x0000000000000000   r9 = 0x0000000000000000  r10 = 0x0000000000000168  r11 = 0x0000000041cb6104  
r12 = 0x0000000000000000  r13 = 0x00007ffd85860a68  r14 = 0x000055fc60a248f8  r15 = 0x00007f0c89d93020  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (<unknown module>) 
==1807243==ABORTING
```

closed 2025-05-06 by calvin2021y

calvin2021y, 2025-05-06:
hi @lum1n0us

with this patch, after enable simd I get this error:

```sh
wamrc -v=5 --invoke-c-api-import --enable-tail-call --format=aot --target=x86_64-pc-linux-gnu --enable-segue -o test.aot test.wasm
wamrc -v=5 --invoke-c-api-import --enable-tail-call --format=aot --target=x86_64-pc-linux-gnu --cpu=skylake -mllvm -mattr=+sse2 --enable-segue -o db0.aot db0.wasm

[15:15:50:699 - 7F12D1949780]: triple: x86_64-pc-linux-gnu => normailized: x86_64-pc-linux-gnu
Create AoT compiler with:
  target:        x86_64
  target cpu:    
  target triple: x86_64-pc-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Error: SIMD compatibility check failed, try adding --cpu=<cpu> to specify a cpu or adding --disable-simd to disable SIMD
wamrc return                                     time of last stage: 0 ms, total time: 7 ms
```

calvin2021y, 2025-05-06:
```sh
wamrc -v=5 --invoke-c-api-import --enable-tail-call --format=aot --target=x86_64-apple-darwin --enable-segue -o test.aot test.wasm

Begin to create compile context                  time of last stage: 7 ms, total time: 7 ms
[15:33:08:378 - 7F40DD6D7780]: triple: x86_64-apple-darwin => normailized: x86_64-apple-darwin
Create AoT compiler with:
  target:        x86_64
  target cpu:    
  target triple: x86_64-apple-darwin
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Begin to compile                                 time of last stage: 8 ms, total time: 15 ms
Begin to compile WASM bytecode to LLVM IR        time of last stage: 0 ms, total time: 15 ms
Begin to verify LLVM module                      time of last stage: 136 ms, total time: 151 ms
Begin to run llvm optimization passes            time of last stage: 54 ms, total time: 205 ms
Finish llvm optimization passes                  time of last stage: 6273 ms, total time: 6478 ms
Begin to emit AOT file                           time of last stage: 0 ms, total time: 6478 ms
Begin to emit object file to buffer              time of last stage: 0 ms, total time: 6478 ms
Begin to emit object file                        time of last stage: 0 ms, total time: 6478 ms
Begin to resolve object file info                time of last stage: 6266 ms, total time: 12744 ms
Error: invalid llvm binary bin_type.
wamrc return                                     time of last stage: 78 ms, total time: 12822 ms
```

`wamrc -v=5 --invoke-c-api-import --enable-tail-call --format=aot --target=x86_64-apple-darwin` , 
 `wamrc -v=5 --invoke-c-api-import --enable-tail-call --format=aot --target=arm64-apple-darwin`, same results.