wasm-micro-runtime/wasm-micro-runtime #4130: In Windows Platform AOT Mode, What Are the Best Practices for Handling the Missing Internal Function __chkstk?

lxbxl, 2025-03-06:
When a wasm app has large stack usage, the AOT compiled file generated by wamrc will insert calls to __chkstk, leading to an error "resolve symbol __chkstk failed" in iwasm.

[This function is not defined in an SDK header and must be declared by the caller. This function is exported from kernelbase.dll.](https://learn.microsoft.com/en-us/windows/win32/devnotes/-win32-__chkstk)

It might be resolved by using` wasm_native_register_natives` or adding a new symbol via `SymbolMap target_sym_map`. However, considering that the calling convention for __chkstk as seen in chkstk.obj is __fastcall, can this calling convention be incorporated into the Native API or target_sym_map?

Here is the decompiled code of chkstk.obj in IDA:
```c
unsigned __int64 __fastcall _chkstk()
{
  unsigned __int64 result; // rax
  char *v1; // r10
  char *StackLimit; // r11
  char v3; // [rsp+18h] [rbp+8h] BYREF
  v1 = &v3 - result;
  if ( (unsigned __int64)&v3 < result )
    v1 = 0i64;
  StackLimit = (char *)NtCurrentTeb()->NtTib.StackLimit;
  if ( v1 < StackLimit )
  {
    LOWORD(v1) = (unsigned __int16)v1 & 0xF000;
    do
    {
      StackLimit -= 4096;
      *StackLimit = 0;
    }
    while ( v1 != StackLimit );
  }
  return result;
}
```

```asm
                public __chkstk
__chkstk        proc near               ; DATA XREF: .pdata:0000000000000068↓o

var_10          = qword ptr -10h
var_8           = qword ptr -8
arg_0           = byte ptr  8

                sub     rsp, 10h        ; _alloca_probe
                mov     [rsp+10h+var_10], r10
                mov     [rsp+10h+var_8], r11
                xor     r11, r11
                lea     r10, [rsp+10h+arg_0]
                sub     r10, rax
                cmovb   r10, r11
                mov     r11, gs:10h
                cmp     r10, r11
                jnb     short cs20
                and     r10w, 0F000h

cs10:                                   ; CODE XREF: __chkstk+3E↓j
                lea     r11, [r11-1000h]
                mov     byte ptr [r11], 0
                cmp     r10, r11
                jnz     short cs10
cs20:                                   ; CODE XREF: __chkstk+28↑j
                mov     r10, [rsp+10h+var_10]
                mov     r11, [rsp+10h+var_8]
                add     rsp, 10h
                retn
__chkstk        endp     
```
If needed, I also have the assembly files chkstk.asm for WinXP and Win2003.

My questions are:

1. What are the official best practices for addressing the "resolve symbol __chkstk failed" error?
2. How can APIs with calling conventions such as __fastcall or __stdcall be invoked in the form of Native APIs?

lum1n0us, 2025-03-12:
Could you please provide a .wasm and .aot file for reproduction purposes?

lxbxl, 2025-03-12:
OK.

[chkstk-case.zip](https://github.com/user-attachments/files/19204729/chkstk-case.zip)

main.c is the source code, and despite not being a large stack example, it still has the __chkstk function generated.  

build AOT cmd:
```bash
./wamrc.exe --format=aot --opt-level=0 -v=5 --target-abi=msvc --cpu=x86-64 -o WamrApp.aot WamrApp.wasm
```