wasm-micro-runtime/wasm-micro-runtime #4134: [RFC] Supporting reference types in struct fields in AOT mod

lum1n0us, 2025-03-11:
This is to track the need for supporting reference types in struct fields when in AOT mode. At present, only primitive types are handled correctly during the wasm-to-AOT compilation process. References need to be reconstructed in the AOT world rather than reusing the results from the wasm world.

Below is a code snippet that illustrates this requirement, though it does not encompass all scenarios.

```wat
(module
  (type $f1 (func (param i32) (result i32)))

  (type $vt1
    (sub
      (struct
        (field (ref null $f1))
      )
    )
  )

  (global $g1 (ref $vt1)
    ref.func $f1   ;; // <- TBD
    struct.new $vt1
  )

  (func $f1 (type $f1) (param i32) (result i32)
    i32.const 0
  )

  (func (export "run") (result i32)
    global.get $g1
    struct.get $vt1 0
    ref.test (ref null $f1)  ;; // should be 1 as true
  )
)

```

chenliuyang1989, 2025-03-12:
Thanks~！Excuse me, when can it be finished? An approximate time is ok

xujuntwt95329, 2025-03-31:
We have identified the root cause and decided the fix solution, the patch is still WIP and the estimated finish time is later this week, thanks for your patience.