wasm-micro-runtime/wasm-micro-runtime #4137: Failed to build iwasm with WAMR_BUILD_WASI_NN enabled

zhen9910, 2025-03-13:
### Subject of the issue

On Linux, it is failed to build iwasm with WAMR_BUILD_WASI_NN enabled.
This works on WAMR-2.2.0 release, but failed on the latest commit.

### Test case
build iwasm for linux with WAMR_BUILD_WASI_NN enabled

### Your environment

* Host OS: Ubuntu 22.04 on X86

### Steps to reproduce
```
$ cd ./product-mini/platforms/linux

$ cmake -S . -B build     -DWAMR_BUILD_WASI_NN=1 -DWAMR_BUILD_WASI_EPHEMERAL_NN=1 -DWAMR_BUILD_WASI_NN_TFLITE=1

$ cmake --build build -j8
```
It failed with error llibiwasm is missing
```
[ 14%] Linking CXX shared library libwasi_nn_tflite.so
/usr/bin/ld: cannot find -llibiwasm: No such file or directory
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/wasi_nn_tflite.dir/build.make:177: libwasi_nn_tflite.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1382: CMakeFiles/wasi_nn_tflite.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```
### ANALYSIS

This build was succeeded on WAMR-2.2.0 release, but failed on the latest one.

It looks a recent change https://github.com/bytecodealliance/wasm-micro-runtime/pull/4076 on `./product-mini/platforms/linux/CMakeLists.txt` renamed libiwasm to vmlib, but [wasi-nn.cmake](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/wasi-nn/cmake/wasi_nn.cmake#L42) still wants to link `libiwasm.so`

I tried a work around below in `./product-mini/platforms/linux/CMakeLists.txt` to add back `libiwasm.so` and can resolve the issue. Please review if this is a valid solution or provide a formal one.
```
add_library (libiwasm SHARED ${WAMR_RUNTIME_LIB_SOURCE})
install (TARGETS libiwasm DESTINATION lib)
set_target_properties (libiwasm PROPERTIES OUTPUT_NAME iwasm)
set_version_info (libiwasm)
target_link_libraries (libiwasm ${LLVM_AVAILABLE_LIBS} ${UV_A_LIBS} -lm -ldl -lpthread)
```

lum1n0us, 2025-03-13:
I would prefer to modify *wasi-nn.cmake*. It's less work. Simply substituting `libiwasm` with `vmlib` should suffice.

zhen9910, 2025-03-13:
@lum1n0us yes, thanks, this change is better! Upload a PR for review. https://github.com/bytecodealliance/wasm-micro-runtime/pull/4138

closed 2025-03-14 by lum1n0us