wasm-micro-runtime/wasm-micro-runtime #4141: WAMR output is different from other runtimes

teyahb8, 2025-03-15:
### Subject of the issue

I have generated a wasm binary as a test case which I ran with wasmtime, wasmer, and wamr (AOT mode) runtimes. The wamr runtime is producing a different output compared to the other runtimes. After shrinking and debugging, the issue seems to arise from nan propagation to the `i64.reinterpret_f64` instruction?

### Test case

```
(module
  (func (;0;) (export "main") (result i64)
    f64.const -nan:0xfffffffffbfff (;=-nan;)
    f64.const 0x1.18475d3690e89p+985 (;=3.58011e+296;)
    f64.max
    i64.reinterpret_f64
  )
)
```

### Your environment

* Host OS: Linux
* WAMR version 2.2.0, Linux, x86_64, AOT mode

### Steps to reproduce

1. Run `wat2wasm file.wat`
2. Run `wamrc --bounds-checks=1 -o file.aot file.wasm`
3. Run `iwasm -f main file.aot`

### Expected behavior

Output from wasmtime and wasmer: 
```-16385```

### Actual behavior

Output from wamr:
`0x7ff8000000000000:i64` which should be equal to: `9221120237041090560`

Can you please confirm if this looks like a bug? Thanks in advance.

TianlongLiang, 2025-03-17:
This is not a bug, the root casue is the operation on NaN. It's similar to previous issue https://github.com/bytecodealliance/wasm-micro-runtime/issues/2833. Other runtimes are using qNaN(0x7fc00000) as the result while WAMR is using 0xffffffff(it is a NaN with arbitrary content) as the result. It is a valid behavior in Wasm spec. Then WAMR operates following instructions in a different way.

teyahb8, 2025-03-17:
Thanks for the clarification. So, can we consider that #4139 and #4140 are happening for the same reason?

TianlongLiang, 2025-03-17:
Yes, I think those two are due to the same reason

teyahb8, 2025-03-17:
I understand. Thanks for clarifying. I will close the issues.

closed 2025-03-17 by teyahb8