wasm-micro-runtime/wasm-micro-runtime #4146: compile error in main branch

king729, 2025-03-19:
when i compile main branch i got compile error
/usr/bin/ld: libvmlib.a(wasm_runtime_common.c.o):(.data.rel.invokeNative_Void+0x0): undefined reference to `invokeNative'

TianlongLiang, 2025-03-19:
Hi, could you please provide more details? Like what platform and architecture you are on?

king729, 2025-03-19:
build target architecture is x86_64  on x86_64 platform

> Hi, could you please provide more details? Like what platform and architecture you are on?

TianlongLiang, 2025-03-19:
What OS and what toolchain(compiler) you are using?

king729, 2025-03-19:
> What OS and what toolchain(compiler) you are using?

[image]
OS is ubuntu20.04 compiler is gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)

TianlongLiang, 2025-03-19:
Can you check your build log to see whether there exist something like:
```bash
[ 78%] Building ASM object CMakeFiles/vmlib.dir/home/tl/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64_simd.s.o
```
It seems to me that your build somehow don't include the asm file. It would be helpful if you could attach the cmake configuration log generated by:
```bash
cmake -B build
```
My machine is nearly the same to you, same architecture same OS same toolchain, and it print something like:
```bash
 cmake -B build                         
-- WAMR version: 2.2.0
-- Build Configurations:
     Build as target X86_64
     Build for platform linux
     CMAKE_BUILD_TYPE Release
     BUILD_SHARED_LIBS OFF
     WAMR Interpreter enabled
     Fast interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin enabled
     Libc WASI enabled
     Multiple modules disabled
     Copy callstack disabled
     Wakeup of blocking operations enabled
     GC performance profiling disabled
     Module instance context enabled
     Write linear memory base addr to x86 GS register enabled
     Quick AOT/JIT entries enabled
     AOT intrinsics enabled
     Shrunk memory enabled
-- About Wasm Proposals:
     Always-on:
       "Extended Constant Expressions"
       "Multi-value"
       "Non-trapping float-to-int conversions"
       "Sign-extension operators"
       "WebAssembly C and C++ API"
     Configurable. 0 is OFF. 1 is ON:
       "Bulk Memory Operation" via WAMR_BUILD_BULK_MEMORY: 1
       "Fixed-width SIMD" via WAMR_BUILD_SIMD: 1
       "Garbage collection" via WAMR_BUILD_GC: 0
       "Legacy Exception handling" via WAMR_BUILD_EXCE_HANDLING: 0
       "Memory64" via WAMR_BUILD_MEMORY64: 0
       "Multiple memories" via WAMR_BUILD_MULTI_MEMORY: 0
       "Reference Types" via WAMR_BUILD_REF_TYPES: 1
       "Reference-Typed Strings" via WAMR_BUILD_STRINGREF: 0
       "Tail call" via WAMR_BUILD_TAIL_CALL: 0
       "Threads" via WAMR_BUILD_SHARED_MEMORY: 0
       "Typed Function References" via WAMR_BUILD_GC: 0
     Unsupported (>= Phase4):
       "Branch Hinting"
       "Custom Annotation Syntax in the Text Format"
       "Exception handling"
       "Import/Export of Mutable Globals"
       "JS String Builtins"
       "Relaxed SIMD"

-- Configuring done (0.1s)
-- Generating done (0.0s)
```

king729, 2025-03-19:
thereis no Building ASM object CMakeFiles/vmlib.dir/home/tl/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64_simd.s.o  in my compile

i just switch to WAMR-2.2.0 tag and it is ok

closed 2025-03-21 by king729