wasm-micro-runtime/wasm-micro-runtime #4166: Question: Is it possible to statically link AOT modules into the host program?

gechdcb, 2025-03-28:
Hi, I tested WAMR on iOS, and the interpreter mode worked fine. 
   I know iOS does not support using mmap to allocate executable memory, so AOT mode is not supported on iOS. 
   So my question is, is it possible to static link the AOT module into the host program, and then load the AOT module? After all, wamrc can output native object format.

TianlongLiang, 2025-03-31:
If I remember correctly, in iOS, the mmap can't mmap a region of memory that is both writable and executable at the same time, but it can use mprotect to change its state later. So maybe for AOT loading, it can load the code first and at the end of the loading phase make it executable

TianlongLiang, 2025-03-31:
You are right that the App Store policy wouldn't allow it. I was speaking from a purely technical standpoint, and it would only work for development.

TianlongLiang, 2025-03-31:
For your original question:
> is it possible to static link the AOT module into the host program
It's possible, and in this case, I think you don't even have to use runtime. What you are trying to achieve is more like this project: https://github.com/web-devkits/wasm2native

gechdcb, 2025-03-31:
Yes, there are other wasm runtimes that can statically link AOT modules, but my use case is a bit different. What I'm trying to do is convert part of a wasm module to AOT, while leaving other parts for the interpreter to execute. Not many runtimes can support this.

kisg, 2025-04-05:
We implemented a similar solution in MOE 1.x: https://multi-os-engine.org . The AOT compiled Dalvik bytecode is packaged into a separate segment in the Mach-O binary, and then remapped as executable on startup. The whole Mach-O binary is codesigned. Such apps can be distributed in the App Store without problems.

lengye7, 2025-04-05:
@TianlongLiang 
Can https://github.com/web-devkits/wasm2native be used on IOS App?

TianlongLiang, 2025-04-10:
In theory it should work, but I didn't try it on iOS platform and it may need some extra configuration

lengye7, 2025-04-10:
OK,thanks for your answer.