wasm-micro-runtime/wasm-micro-runtime #4186: Ubuntu 20.04 LTS (CI) runner will be removed

lum1n0us, 2025-04-08:
> This is a scheduled Ubuntu 20.04 brownout. Ubuntu 20.04 LTS runner will be removed on 2025-04-15. For more details, see https://github.com/actions/runner-images/issues/11101

lum1n0us, 2025-04-08:
ubuntu-20.04 are used in  https://github.com/search?q=repo%3Abytecodealliance%2Fwasm-micro-runtime+ubuntu-20.04+path%3A%2F%5E%5C.github%5C%2Fworkflows%5C%2F%2F&type=code

- [.github/workflows/release_process.yml](https://github.com/bytecodealliance/wasm-micro-runtime/blob/4e50d2191ca8f177ad03a9d80eebc44b59a932db/.github/workflows/release_process.yml#L68)
- [.github/workflows/coding_guidelines.yml](https://github.com/bytecodealliance/wasm-micro-runtime/blob/4e50d2191ca8f177ad03a9d80eebc44b59a932db/.github/workflows/coding_guidelines.yml#L22)
- [.github/workflows/build_llvm_libraries.yml](https://github.com/bytecodealliance/wasm-micro-runtime/blob/4e50d2191ca8f177ad03a9d80eebc44b59a932db/.github/workflows/build_llvm_libraries.yml#L98)
- [.github/workflows/nightly_run.yml](https://github.com/bytecodealliance/wasm-micro-runtime/blob/4e50d2191ca8f177ad03a9d80eebc44b59a932db/.github/workflows/nightly_run.yml#L57)
- [.github/workflows/compilation_on_sgx.yml](https://github.com/bytecodealliance/wasm-micro-runtime/blob/4e50d2191ca8f177ad03a9d80eebc44b59a932db/.github/workflows/compilation_on_sgx.yml#L67)

yamt, 2025-04-18:
we no longer use ubuntu-20.04 after https://github.com/bytecodealliance/wasm-micro-runtime/pull/4191. let's close.

closed 2025-04-18 by yamt