wasm-micro-runtime/wasm-micro-runtime #4188: WAMR aot mode crash on Zephyr

banana-sun, 2025-04-10:
* Host system: Zephyr v3.7.1
* WAMR version: dev/zephyr_file_socket
* Hardware: ST NUCLEO-H563ZI

```cmd
Build Configurations:
     Build as target THUMBV8
     CMAKE_BUILD_TYPE 
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin disabled
     Libc WASI enabled
     Fast interpreter disabled
     Multiple modules disabled
     Bulk memory feature enabled
     Shared memory enabled
     Thread manager enabled
     Lib pthread enabled
     Wakeup of blocking operations enabled
     Reference types enabled
     GC performance profiling disabled
     Global heap pool enabled
     Custom global heap size: 262144
     Module instance context enabled
     Quick AOT/JIT entries enabled
     AOT intrinsics enabled
```

It supports both `aot` and `interpreter` modes. Both of the following files work properly when using interpreter mode. However, when an aot file is generated using the following command, the target file of file 2 will crash and the target file of file 1 run normally.

```cmd
wamrc --target=thumbv8m.main --cpu=cortex-m33 -o <test_file.aot> <test_file.wasm>
```

Error output:

```txt
[00:01:18.138,000] <err> os: ***** USAGE FAULT *****
[00:01:18.138,000] <err> os:   No coprocessor instructions
[00:01:18.138,000] <err> os: r0/a1:  0x20006e18  r1/a2:  0x00000000  r2/a3:  0x0000046a
[00:01:18.138,000] <err> os: r3/a4:  0x00005478 r12/ip:  0x00005450 r14/lr:  0x2001cea3
[00:01:18.138,000] <err> os:  xpsr:  0x210d0000
[00:01:18.138,000] <err> os: Faulting instruction address (r15/pc): 0x2001d142
[00:01:18.138,000] <err> os: >>> ZEPHYR FATAL ERROR 33: Unknown error on CPU 0
[00:01:18.138,000] <err> os: Current thread: 0x20002fa0 (unknown)
[00:01:18.212,000] <err> os: Halting system
```

Test app file 1:

```c
#include <stdio.h>
#include <stdlib.h>

#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <unistd.h>
#include <string.h>
#include <errno.h>
#include <dirent.h>

static int file_test(void)
{
    int     fd;
    uint8_t buf[32] = { 0 };

    fd = open("/lfs/test.txt", O_CREAT | O_RDWR);
    if (fd < 0)
    {
        printf("Error opening file [%d]\n", fd);
        return fd;
    }

    close(fd);
    return 0;
}

int main(int argc, char **argv)
{
    printf("Hello, I'm WASM application\r\n");

    file_test();
    return 0;
}
```

Test file 2:

```c
#include <stdio.h>
#include <stdlib.h>

#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <unistd.h>
#include <string.h>
#include <errno.h>
#include <dirent.h>

static int file_test(void)
{
    int     fd;
    uint8_t buf[32] = { 0 };

    fd = open("/lfs/test.txt", O_CREAT | O_RDWR);
    if (fd < 0)
    {
        printf("Error opening file [%d]\n", fd);
        return fd;
    }

    uint32_t len = read(fd, buf, sizeof(buf));
    if (len < 0)
    {
        printf("Error reading file [%d]\n", len);
        return len;
    }
    printf("Read: %s\n", buf);

    close(fd);
    return 0;
}

int main(int argc, char **argv)
{
    printf("Hello, I'm WASM application\r\n");

    file_test();
    return 0;
}
```

banana-sun, 2025-04-16:
@xujuntwt95329 Thanks for your support. I have found the cause of the problem.

By looking at the assembly file of the aot file, I found that floating-point arithmetic instructions were used in the following processing. But I don't enable FPU on Zephyr.

```c
printf("Read: %s\n", buf);
```

When I add *CONFIG_FPU=y* at `prj.conf` file, the issue is solved.

banana-sun, 2025-04-16:
This issue has been solved, I think it can be closed. :-)

lum1n0us, 2025-04-16:
If it's not too much trouble, I hope you can document and summarize a .md on how to port WAMR to Zephyr+<board name>. It will be helpful.

banana-sun, 2025-04-17:
I'm very willing to do so. But for now, I'm doing it based on [this step](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/zephyr/simple/README.md) by **Local Environment**. At present, my energy is mainly focused on researching the capabilities of Zephyr + WAMR.

About how to port WAMR to Zephyr, I also expect an official method at the same time. Of course, there doesn't seem to be a very good method now. I think we can find a good way to port together. If necessary, I'll be very glad to help do something.

closed 2025-04-22 by lum1n0us