wasm-micro-runtime/wasm-micro-runtime #4190: [gc] array as a field in struct object is not supported in constant init expr

vouillon, 2025-04-10:
### Steps to reproduce

1. Unpack zip [globals.zip](https://github.com/user-attachments/files/18892034/globals.zip)

2. Build `iwasm` with the following command:
```
cmake .. -DWAMR_BUILD_GC=1 -DWAMR_BUILD_EXCE_HANDLING=1 -DWAMR_BUILD_TAIL_CALL=1 && make
```

3.  Run `./iwasm globals.wasm`

### Expected behavior

It should print these three lines:
```
line 1
line 2
line 3
```

### Actual behavior

I get the following output:
```
WASM module instantiate failed: array as a field in struct object is not supported in constant init expr
```

nickbetteridge, 2025-04-11:
Experiencing the same issue - getting a `unhandled SIGSEGV, si_addr: 0x642700000005` with the above WASM error

nickbetteridge, 2025-04-24:
Is there any chance of somebody being able to take a few moments to have a look at this? This is a bit of a show-stopper for us and, unfortunately, I'm not a C/C++ coder - so we're a little stuck as we can't, currently, build and run anything via Ocaml. Thanks in advance for any feedback