wasm-micro-runtime/wasm-micro-runtime #4194: inconsistent cmake version requirements

yamt, 2025-04-14:
with https://github.com/bytecodealliance/wasm-micro-runtime/pull/4076, we started using [CMAKE_CURRENT_FUNCTION_LIST_DIR](https://cmake.org/cmake/help/latest/variable/CMAKE_CURRENT_FUNCTION_LIST_DIR.html), which requires cmake 3.17.

on the other hand, many of our CMakeLists.txt have cmake_minimum_required with 3.14.

lum1n0us, 2025-04-14:
Which option do you prefer?

- Updating `cmake_minimum_required()` in all CMakeLists.txt to 3.17. (we can make it happen with #4180) 
- Or replacing `CMAKE_CURRENT_FUNCTION_LIST_DIR` with an alternative?

yamt, 2025-04-14:
> Which option do you prefer?
> 
>     * Updating `cmake_minimum_required()` in all CMakeLists.txt to 3.17. (we can make it happen with [Keep fix the CMake compatibility issue #4180](https://github.com/bytecodealliance/wasm-micro-runtime/pull/4180))
> 
>     * Or replacing `CMAKE_CURRENT_FUNCTION_LIST_DIR` with an alternative?

i prefer the latter if it can be done in a reasonable way.

closed 2025-04-16 by lum1n0us