wasm-micro-runtime/wasm-micro-runtime #4195: wasm_runtime_start_debug_instance can break the module

yamt, 2025-04-14:
wasm_runtime_start_debug_instance uses wasm_runtime_module_malloc to reserve memory for [_M](https://lldb.llvm.org/resources/lldbgdbremote.html#m-size-permissions).
it isn't safe because the in-module heap might have not been initialized properly yet.

yamt, 2025-04-14:
a simple workaround for those who don't really need expression evaluation:

```
diff --git a/core/iwasm/libraries/debug-engine/debug_engine.c b/core/iwasm/libraries/debug-engine/debug_engine.c
index 0ffc78ad..1b81a41c 100644
--- a/core/iwasm/libraries/debug-engine/debug_engine.c
+++ b/core/iwasm/libraries/debug-engine/debug_engine.c
@@ -408,6 +408,8 @@ wasm_debug_instance_create(WASMCluster *cluster, int32 port)
      * the allocation failed, the debugger will not be able to evaluate
      * expressions */
     instance->exec_mem_info.size = DEBUG_EXECUTION_MEMORY_SIZE;
+    instance->exec_mem_info.start_offset = 0;
+#if 0
     instance->exec_mem_info.start_offset = wasm_runtime_module_malloc(
         module_inst, (uint64)instance->exec_mem_info.size, NULL);
     if (instance->exec_mem_info.start_offset == 0) {
@@ -417,6 +419,7 @@ wasm_debug_instance_create(WASMCluster *cluster, int32 port)
             "Will not be able to evaluate expressions during "
             "debugging");
     }
+#endif
     instance->exec_mem_info.current_pos = instance->exec_mem_info.start_offset;
 
     if (!wasm_debug_control_thread_create(instance, port)) {
```

yamt, 2025-04-16:
> > it isn't safe because the in-module heap might have not been initialized properly yet.
> 
> like calling `wasm_runtime_start_debug_instance_with_port()` before `wasm_runtime_instantiate()`?

no.

eg. call `wasm_runtime_start_debug_instance_with_port()` before `wasm_application_execute_main()`.

lum1n0us, 2025-04-17:
not entirely clear on this. Could you provide us with more details? As we understand it, after `wasm_runtime_instantiate()`, the linear memory, including the heap area, has been initialized, and if it's a wasm32-wasip1, all `ctor()` has also been executed(by `execute_post_instantiate_functions()`). 

This would be unless there are exported wasm functions that will call `memory.grow()`?

lum1n0us, 2025-04-17:
Just my wide guess. Please provide us with more details?

yamt, 2025-04-17:
malloc/free are not expected to be usable before _start.

well, of course, it's possible to implement malloc as it can be used before _start.
however, we should not assume such implementations in general.

lum1n0us, 2025-04-17:
if *in-module heap preparation* is about below in [`_start()`](https://github.com/WebAssembly/wasi-libc/blob/e9524a0980b9bb6bb92e87a41ed1055bdda5bb86/libc-bottom-half/crt/crt1-command.c#L11) ,

``` c
// in void _start(void) 
#ifdef _REENTRANT
    __wasi_init_tp();
#endif

    // The linker synthesizes this to call constructors.
    __wasm_call_ctors();

    // goes to __main_void();
```

and seems they are only called from `_start()`. Even another export function will not execute them. 

So if users want to execute an exported function, like via `iwasm -f foo`, the `__wasm_call_ctors()` won't be executed, because `_start()` is passed (correct me if I am wrong). Is that correct when we believe `__wasm_call_ctors()` is important?

yamt, 2025-04-18:
> if _in-module heap preparation_ is about below in [`_start()`](https://github.com/WebAssembly/wasi-libc/blob/e9524a0980b9bb6bb92e87a41ed1055bdda5bb86/libc-bottom-half/crt/crt1-command.c#L11) ,
> 
> // in void _start(void) 
> #ifdef _REENTRANT
>     __wasi_init_tp();
> #endif
> 
>     // The linker synthesizes this to call constructors.
>     __wasm_call_ctors();
> 
>     // goes to __main_void();
> 
> and seems they are only called from `_start()`. Even another export function will not execute them.
> 
> So if users want to execute an exported function, like via `iwasm -f foo`, the `__wasm_call_ctors()` won't be executed, because `_start()` is passed (correct me if I am wrong).

my understanding is that extra exported functions are not expected to be called before or after _start.
they can be called during the execution of _start. (eg. host functions called by _start, like wasi, can call them back)

> Is that correct when we believe `__wasm_call_ctors()` is important?

usually, if a module exports _start, it doesn't export __wasm_call_ctors.

i guess there were historical conventions using exported __wasm_call_ctors. but i'm not familiar with them.
i suspect our logic in execute_post_instantiate_functions was for such legacy conventions.
wasip1 commands don't export __wasm_call_ctors.

lum1n0us, 2025-04-22:
> they can be called during the execution of _start. (eg. host functions called by _start, like wasi, can call them back)

May I ask how?

> it isn't safe because the in-module heap might have not been initialized properly yet.

It seems there isn't a good method to permit the host to call `wasm_malloc()` after the in-module heap has been properly initialized by `_start()`. (This issue might be lessened if dealing with a reactor .wasm module)

yamt, 2025-04-22:
> > they can be called during the execution of _start. (eg. host functions called by _start, like wasi, can call them back)
> 
> May I ask how?

wasip1 itself doesn't have such an api.

wasi-threads uses `wasi_thread_start` export for thread entry point.

> It seems there isn't a good method to permit the host to call `wasm_malloc()` after the in-module heap has been properly initialized by `_start()`. (This issue might be lessened if dealing with a [...]

a reactor doesn't have the same problem as we call its entry point `_initialize` during instantiation.

yamt, 2025-04-22:
anyway, currently this isn't high-proirity (for me) because the default malloc implementation of wasi-libc seems to be safe to be used before _start.
but it might hurt you eg. when you update libc.

lum1n0us, 2025-04-22:
Certainly. I will bear this in mind and continue to monitor any upcoming changes to wasi-libc.