wasm-micro-runtime/wasm-micro-runtime #4206: SIMD support on windows?

unvestigate, 2025-04-19:
I recently updated from a version of main branch from Feb 2025 to the latest one, and I found all parameters to be null/0 when calling a native function from WASM.

When doing a diff between the versions I saw that the following text had been added to build_wamr.md: "When enabling SIMD for fast interpreter mode, you'll need to enable both SIMD and the SIMDe library". However, if I enabled it CMake tells me that "SIMDe doesnt support platform windows".

It seems like the only way for me to get my app working is to set WAMR_BUILD_SIMD to 0. What is the plan for SIMD support for windows, going forward?

lum1n0us, 2025-04-19:
> "SIMDe doesnt support platform windows" 

That's true. 

> What is the plan for SIMD support for windows, going forward?

@loganek @jammar1  is there any plan for enabling SIMD for fast interpreter mode for windows?

unvestigate, 2025-04-19:
Also, what was the behavior before? Was SIMD ever working for windows using the fast interpreter? These are the CMake toggles I used with the previous version:

set(WAMR_BUILD_PLATFORM "windows")
set(WAMR_BUILD_TARGET "X86_64")
set(WAMR_BUILD_INTERP 1)
set(WAMR_BUILD_FAST_INTERP 1)
set(WAMR_BUILD_AOT 1)
set(WAMR_BUILD_LIBC_BUILTIN 1)
set(WAMR_BUILD_LIBC_WASI 1)
set(WAMR_BUILD_SIMD 1)
set(WAMR_BUILD_REF_TYPES 1)
set(WAMR_BUILD_JIT 0)
set(WAMR_BUILD_FAST_JIT 0)

I had AOT enabled, but didn't end up using it. There was some issue with it that I don't remember right now. Regardless, here I had WAMR_BUILD_SIMD enabled and at least my parameter values were correct, rather than all zero.

lum1n0us, 2025-04-19:
> Was SIMD ever working for windows using the fast interpreter?

Not until @jammar1 contributes that feature.

jammar1, 2025-04-19:
When I tried it myself on Windows it built and ran successfully. I think the SIMD spec tests passed too but I would have to check. I suspect the only issue is related to CI configuration and it is probably quite simple to fix.

From what I remember it was having problems building SIMDe with C99 compound literals (e.g https://github.com/jammar1/wasm-micro-runtime/actions/runs/13629969332/job/38095386779#step:4:350).

And I remember SGX didn't work because the toolchain was missing headers for AVX intrinsics (e.g the AVX2 macros were defined, but the headers were not available).

I will have a look at it next week.

It's concerning if "all parameters to be null/0" though.

unvestigate, 2025-04-19:
Yeah, it's weird. Parameters of WASM functions called from native seem to work fine, as well as return values from those functions. But parameters to native functions called from WASM are all zero. Let me know how this proceeds. I might be able to provide you with a reproducing test case if needed.

unvestigate, 2025-04-19:
Even the first, implicit parameter `wasm_exec_env_t exec_env`, is zero.

jammar1, 2025-04-19:
Could you try building with SIMDe/deleting the Windows platform check at build-scripts/runtime_lib.cmake, line 159 and seeing if that resolves the issue?

unvestigate, 2025-04-19:
Yes, that seems to work around the issue. I commented out the check like this:

```
if (WAMR_BUILD_SIMD EQUAL 1 AND WAMR_BUILD_FAST_INTERP EQUAL 1)
    #if (WAMR_BUILD_PLATFORM STREQUAL "windows")
    #    message(STATUS "SIMDe doesnt support platform " ${WAMR_BUILD_PLATFORM})
    #    set(WAMR_BUILD_SIMDE 0)
    #else()
        include (${IWASM_DIR}/libraries/simde/simde.cmake)
        set (WAMR_BUILD_SIMDE 1)
    #endif()
endif ()
```

jammar1, 2025-04-19:
That's good, I will try and fix the Windows CI issues next week then.

lum1n0us, 2025-04-24:
@jammar1 Needs to disable simde from SGX.

```shell
In file included from /workspaces/wasm-micro-runtime/product-mini/platforms/linux-sgx/build/_deps/simde-src/simde/wasm/../simde-common.h:45,
                 from /workspaces/wasm-micro-runtime/product-mini/platforms/linux-sgx/build/_deps/simde-src/simde/wasm/simd128.h:31,
                 from /workspaces/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:25:
/workspaces/wasm-micro-runtime/product-mini/platforms/linux-sgx/build/_deps/simde-src/simde/wasm/../simde-features.h:319:12: fatal error: emmintrin.h: No such file or directory
  319 |   #include <emmintrin.h>
      |            ^~~~~~~~~~~~~
compilation terminated.
```

jammar1, 2025-04-24:
@lum1n0us Regarding Windows, `set(WAMR_BUILD_SIMD 0)` would also disable SIMD in AOT mode, not sure if that's desirable? And I'll add SGX to that if statement.

unvestigate, 2025-04-24:
Well, ideally we'd have SIMD in AOT mode, if possible.

lum1n0us, 2025-04-25:
That's a workaround for [this issue](https://github.com/bytecodealliance/wasm-micro-runtime/issues/4216). Let's move to that issue and discuss a well-balanced patch.