wasm-micro-runtime/wasm-micro-runtime #4210: llvm versions we use doesn't build with the latest Xcode

yamt, 2025-04-24:
llvm (lldb) versions used by wamr doen't build with latest Xcode
because Xcode dropped std:char_traits<unsigned char>:
https://developer.apple.com/documentation/xcode-release-notes/xcode-16_3-release-notes
(search "The base template for std::char_traits has been removed" in the page)

llvm 18 and later has this fixed:
https://github.com/llvm/llvm-project/commit/68744ffbdd7daac41da274eef9ac0d191e11c16d

```shell
cd wamr-compiler
./build_llvm_xtensa.sh --project clang lldb
```

```
In file included from /Volumes/PortableSSD/llvm-xtensa/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/Process/gdb-remote/GDBRemoteCommunicationServerLLGS.cpp:13:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/chrono:1002:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/vector:325:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/formatter_bool.h:19:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/formatter_integral.h:21:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/formatter_output.h:22:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/parser_std_format_spec.h:39:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/string:821:42: error: implicit instantiation of undefined template 'std::char_traits<unsigned char>'
  821 |   static_assert(is_same<_CharT, typename traits_type::char_type>::value,
      |                                          ^
/Volumes/PortableSSD/llvm-xtensa/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/Process/gdb-remote/GDBRemoteCommunicationServerLLGS.cpp:634:34: note: in instantiation of template class 'std::basic_string<unsigned char>' requested here
  634 |       std::basic_string<uint8_t> zeros(reg_info.byte_size, '\0');
      |                                  ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
In file included from /Volumes/PortableSSD/llvm-xtensa/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/Process/gdb-remote/GDBRemoteCommunicationServerLLGS.cpp:13:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/chrono:1001:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/string_view:300:42: error: implicit instantiation of undefined template 'std::char_traits<unsigned char>'
  300 |   static_assert(is_same<_CharT, typename traits_type::char_type>::value,
      |                                          ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__type_traits/is_convertible.h:22:99: note: in instantiation of template class 'std::basic_string_view<unsigned char>' requested here
   22 | struct _LIBCPP_TEMPLATE_VIS is_convertible : public integral_constant<bool, __is_convertible(_T1, _T2)> {};
      |                                                                                                   ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/string:746:29: note: in instantiation of template class 'std::is_convertible<const unsigned int &, std::basic_string_view<unsigned char>>' requested here
  746 |     : public _BoolConstant< is_convertible<const _Tp&, basic_string_view<_CharT, _Traits> >::value &&
      |                             ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/string:1162:27: note: in instantiation of template class 'std::__can_be_converted_to_string_view<unsigned char, std::char_traits<unsigned char>, unsigned int>' requested here
 1162 |             __enable_if_t<__can_be_converted_to_string_view<_CharT, _Traits, _Tp>::value &&
      |                           ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/string:1166:42: note: while substituting prior template arguments into non-type template parameter [with _Tp = uint32_t]
 1166 |   _LIBCPP_CONSTEXPR_SINCE_CXX20 explicit basic_string(const _Tp& __t, const allocator_type& __a)
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1167 |       : __r_(__default_init_tag(), __a) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1168 |     __self_view __sv = __t;
      |     ~~~~~~~~~~~~~~~~~~~~~~~
 1169 |     __init(__sv.data(), __sv.size());
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1170 |   }
      |   ~
/Volumes/PortableSSD/llvm-xtensa/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/Process/gdb-remote/GDBRemoteCommunicationServerLLGS.cpp:634:34: note: while substituting deduced template arguments into function template 'basic_string' [with _Tp = uint32_t, $1 = (no value)]
  634 |       std::basic_string<uint8_t> zeros(reg_info.byte_size, '\0');
      |                                  ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
In file included from /Volumes/PortableSSD/llvm-xtensa/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/Process/gdb-remote/GDBRemoteCommunicationServerLLGS.cpp:13:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/chrono:1002:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/vector:325:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/formatter_bool.h:19:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/formatter_integral.h:21:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/formatter_output.h:22:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__format/parser_std_format_spec.h:39:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/string:2392:3: error: implicit instantiation of undefined template 'std::char_traits<unsigned char>'
 2392 |   traits_type::assign(std::__to_address(__p), __n, __c);
[... 2 lines cut ...]
 1091 |     __init(__n, __c);
      |     ^
/Volumes/PortableSSD/llvm-xtensa/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/Process/gdb-remote/GDBRemoteCommunicationServerLLGS.cpp:634:34: note: in instantiation of member function 'std::basic_string<unsigned char>::basic_string' requested here
  634 |       std::basic_string<uint8_t> zeros(reg_info.byte_size, '\0');
      |                                  ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
In file included from /Volumes/PortableSSD/llvm-xtensa/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/Process/gdb-remote/GDBRemoteCommunicationServerLLGS.cpp:13:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/chrono:1002:
[... cut ...]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/string:2393:3: error: implicit instantiation of undefined template 'std::char_traits<unsigned char>'
 2393 |   traits_type::assign(__p[__n], value_type());
      |   ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.4.sdk/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
4 errors generated.
[2849/5061] Building CXX object tools/...ssGDBRemote.dir/ProcessGDBRemote.cpp.o
ninja: build stopped: subcommand failed.
```

yamt, 2025-04-24:
possible fixes:
a. bump llvm to 18 on later
b. patch llvm locally

lum1n0us, 2025-04-24:
I prefer bump llvm.

yamt, 2025-08-12:
fixed by https://github.com/bytecodealliance/wasm-micro-runtime/pull/4259