wasm-micro-runtime/wasm-micro-runtime #4215: How to deal with "target address out of range." in AOT

AY268, 2025-04-25:
I am trying to use WAMR on Android and built a large AOT file. 

However, I'm encountering a problem with the error message: "AOT module load failed: target address out of range."

This issue occurs in the do_text_relocation function within the aot_loader.c file, at the code block:
```
if (!apply_relocation(
    module, aot_text, aot_text_size, relocation->relocation_offset,
    relocation->relocation_addend, relocation->relocation_type,
    symbol_addr, symbol_index, error_buf, error_buf_size))
```

The values for these parameters are as follows:
[image]

I wonder how to deal with it.

yamt, 2025-04-25:
what version of wamr are you using?
cf. https://github.com/bytecodealliance/wasm-micro-runtime/pull/3743

AY268, 2025-04-27:
> what version of wamr are you using? cf. [#3743](https://github.com/bytecodealliance/wasm-micro-runtime/pull/3743)

I run on the main branch up to commit b6dea221a609606ce6c3dcaa0f4d0399ea5bc4dc, but still have this problem.

yamt, 2025-05-02:
the address of the native function seems too far from aot text for call26.
probably you can use large model (--size-level=0) to work it around.

eloparco, 2025-05-09:
We're hitting the same condition https://github.com/bytecodealliance/wasm-micro-runtime/blob/510bb11f4acad6ecbf1c221b96af9317887770b2/core/iwasm/aot/arch/aot_reloc_arm.c#L362 on many Android devices in production.

Apart from changing the size level (that's not an option for us), are there other mitigations/workarounds? Ideally something on the `wamrc` side

denizsokmen, 2025-05-09:
--enable-indirect-mode seems to be providing a path forward as I don't see JUMP24 relocations being emitted, however it still emits aeabi_ builtins despite disabling llvm intrinsics which may be a problem if PLT is not within the 32MB range. Is there a documentation around AOT format and relocations it uses? How does indirect mode eliminate the relocations?