wasm-micro-runtime/wasm-micro-runtime #4221: bh_read_file.h isn't in any of the include paths.

Please-just-dont, 2025-04-26:
Now can my program link to it. The tutorial has:
```

#include "bh_read_file.h"

buffer = bh_read_file_to_buffer(wasm_path, &buf_size);
```

I cannot include "bh_read_file.h" as it's not in any of the include paths. And if I manually find where it is and include that I get an undefined reference error because it's not linking to it.

In my CMake I do as the instructions say:
```

include (wasm-micro-runtime-main/build-scripts/runtime_lib.cmake)
add_library(vmlib ${WAMR_RUNTIME_LIB_SOURCE})
target_link_libraries(my_target PRIVATE vmlib)
```

How are the include directories even added? I have access to the normal wasm headers like wasm_runtime.h, but not others.

Please-just-dont, 2025-04-26:
I've included everything:

#include "wasm_runtime.h"
#include "wasm_c_api.h"
#include "wasm.h"
#include "wasm_runtime_common.h"
#include "wasm_interp.h"
#include "wasm_memory.h"
#include "wasm_native.h"
#include "wasm_shared_memory.h"
#include "wasm.h"
#include "wasm_export.h"
#include "wasm_exec_env.h"
#include "wasm_export.h"
#include "wasm_loader_common.h"
#include "wasm_loader.h"
#include "wasm_opcode.h"

And still nothing. I also can't find:

`read_wasm_binary_to_buffer`

That's in the runtime initialization documentation.

TianlongLiang, 2025-04-27:
The `bh_read_file.h/.c` is included in `core/shared/utils/uncommon/shared_uncommon.cmake`

Please-just-dont, 2025-04-27:
@TianlongLiang So... I have to include that in my CMake in addition to:

include (wasm-micro-runtime-main/build-scripts/runtime_lib.cmake)
add_library(vmlib ${WAMR_RUNTIME_LIB_SOURCE})
target_link_libraries(my_target PRIVATE vmlib)

???

TianlongLiang, 2025-04-27:
Yes, you can refer to iwasm CMakeLists.txt in product-mini/platforms/linux/CMakeLists.txt
```cmake
include (${SHARED_DIR}/utils/uncommon/shared_uncommon.cmake)

add_executable (iwasm main.c ${UNCOMMON_SHARED_SOURCE})
```

Please-just-dont, 2025-04-27:
@TianlongLiang  Hmm, ok, well it may be a good idea if that were mentioned in the docs as someone else may have trouble. May I ask why it's done this way, including CMakeLists.txt, then you yourself defining a library and then linking to it? As supposed to doing add_subdirectory? Are there two ways to build this? Because I noticed there's a CMakeLists.txt at the root directory.

TianlongLiang, 2025-04-27:
IIUC, what you are trying to do is embed wamr into your project, you can refer to [this document](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/embed_wamr.md#embed-wamr-into-developers-project) and in that section give example about how to write custom CMakeLists.txt to embed wamr, it can used as reference if you want to do custom embed or something: 

"Examples can be found in [CMakeLists.txt of linux platform](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/linux/CMakeLists.txt) and [other platforms](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms)."

TianlongLiang, 2025-04-27:
> As supposed to doing add_subdirectory? Are there two ways to build this? Because I noticed there's a CMakeLists.txt at the root directory.

You can do that too

closed 2025-05-28 by TianlongLiang