wasm-micro-runtime/wasm-micro-runtime #4228: A bug in the instruction `select` on fast interpreter mode

erxiaozhou, 2025-04-28:
### Subject of the issue

According to the specification, the test case output should be the first operand (i.e., <0x0000000000000000 0x0000000000000000>:v128). However, the fast interpreter mode output the wrong result (i.e.,<0x0099055a00000000 0x6e28e1a119d541ac>:v128)

### Test case

```
(module
  (type (;0;) (func))
  (type (;1;) (func (result v128)))
  (func (;0;) (type 1) (result v128)
    v128.const i32x4 0x00000000 0x00000000 0x00000000 0x00000000
    v128.const i32x4 0x8aaed7e8 0x0099055a 0x19d541ac 0x6e28e1a1
    i32.const 2147483647
    select)
  (export "_start" (func 0))
  (export "to_test" (func 0)))
```

### Your environment

* Host OS : Ubuntu 20.04
* WAMR version : 6593b3f34784c0cca1176aec917ba38a22376d5c
* cpu architecture : x86-64
* running mode : fast interpreter mode

### Steps to reproduce

1. transform the given `wat` to `wasm`
2. use the fast interpreter mode to execute it

### Expected behavior

Ourput
```
<0x0000000000000000 0x0000000000000000>:v128
```

### Actual behavior

Ourput
```
<0x0099055a00000000 0x6e28e1a119d541ac>:v128
```

### Extra Info

LLVM JIT mode works well.

kylo5aby, 2025-04-30:
it seems that when `select` return, `EXT_OP_COPY_STACK_TOP_V128` takes two wrong operands. it copy stack top value to addr, but it reversed the `value` and `addr`.  could you take a look @jammar1 

[image]

Zzzabiyaka, 2025-05-02:
a PR to add a test which would catch failures like your one https://github.com/WebAssembly/spec/pull/1899

Thanks for reporting again

PR #4236 (2025-05-14): Add select 128
This PR adds support for select_128 operation which was missing 

I am planning to take an additional action on me to add a corresponding case to spec-test as the current WAMR implementation passed them all 

fixes #4228