wasm-micro-runtime/wasm-micro-runtime #4233: How to enable reference types in the runtime?

Please-just-dont, 2025-04-30:
I can compile code from C++ to WebAssembly using Clang compiler, but if I put a printf in there then WAMR isn't able to load the module because it gives this error:

> 
Load wasm module failed. error: WASM module load failed: zero byte expected. The module uses reference types feature which is disabled in the runtime.

How do I enable reference types to make this work? Log:

> [16:39:15:034 - 7FFFF7E88780]: Load type section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load import section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load function section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load table section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load memory section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load global section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load export section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load table segment section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load code segment section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Load data segment section success.
> 
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_abbrev].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_info].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_str].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_line].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_loc].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_ranges].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_pubnames].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [.debug_pubtypes].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [name].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [producers].
> [16:39:15:034 - 7FFFF7E88780]: Ignore custom section [target_features].
> [16:39:15:034 - 7FFFF7E88780]: /media/moe/14E610DDE610C142/MyProgrammingFiles/Deps/wasm-micro-runtime-main/core/iwasm/common/wasm_runtime_common.c, line 1498, WASM module load failed
> 
> Load wasm module failed. error: WASM module load failed: zero byte expected. The module uses reference types feature which is disabled in the runtime.

Please-just-dont, 2025-04-30:
I've dug through the code and there's a WASM_ENABLE_REF_TYPES flag. Is there any mention of this anywhere so I can read about it? Do I just set the definition for the project and that's it?

lum1n0us, 2025-04-30:
Yes. Switching the compilation flag like `cmake -DWAMR_BUILD_REF_TYPES=1 ...` from the command line or adding `set(WAMR_BUILD_REF_TYPES 1)` in CMakeLists.txt will do the trick.

Please-just-dont, 2025-04-30:
OK, that worked, but now I have the problem:

```
[20:06:09:971 - 7FFFF7E88780]: warning: failed to link import function (wasi_snapshot_preview1, fd_close)
[20:06:09:971 - 7FFFF7E88780]: warning: failed to link import function (wasi_snapshot_preview1, fd_fdstat_get)
[20:06:09:971 - 7FFFF7E88780]: warning: failed to link import function (wasi_snapshot_preview1, fd_seek)
[20:06:09:971 - 7FFFF7E88780]: warning: failed to link import function (wasi_snapshot_preview1, fd_write)
```

And if I enable WASM_ENABLE_MULTI_MODULE I get:
`
can not find a module named wasi_snapshot_preview1 for function fd_close`

I compiled with Clang with option -target wasm32-wasi

lum1n0us, 2025-04-30:
Could you please share your compilation commands for runtime library or iwasm?

Please-just-dont, 2025-04-30:
@lum1n0us  Sure, I link to it using CMake. The CMake file looks like this:

```
set(WAMR_ROOT_DIR ${EXTERNAL_DEPENDENCIES_DIR}/wasm-micro-runtime-main CACHE STRING "")
set(WAMR_BUILD_PLATFORM "linux" CACHE STRING "")

#set(WAMR_BUILD_TARGET) IS AUTOMATICALLY DETECTED BUT YOU MAY NEED TO SET IT
set(WAMR_BUILD_INTERP 1 CACHE BOOL "Enable fast interpreter mode")

set(WAMR_BUILD_FAST_INTERP 1 CACHE BOOL "Enable fast interpreter mode")
set(WAMR_BUILD_SHARED_MEMORY 1 CACHE BOOL "")

include (${WAMR_ROOT_DIR}/build-scripts/runtime_lib.cmake)
include (${WAMR_ROOT_DIR}/core/shared/utils/uncommon/shared_uncommon.cmake)

add_library(vmlib 
${WAMR_RUNTIME_LIB_SOURCE}
${UNCOMMON_SHARED_SOURCE}
)

target_compile_definitions(vmlib 
PRIVATE WASM_ENABLE_REF_TYPES=1
WASM_ENABLE_MULTI_MODULE=1
)

target_link_libraries(${main_target_name} PRIVATE vmlib)
```

lum1n0us, 2025-04-30:
> [20:06:09:971 - 7FFFF7E88780]: warning: failed to link import function (wasi_snapshot_preview1, fd_write)

`set(WAMR_BUILD_LIBC_WASI 1)` in CMakeLists.txt or `-DWAMR_BUILD_LIBC_WASI=1` in commoand line

Please-just-dont, 2025-04-30:
@lum1n0us Awesome, thank you. It's taking me ages but I'm slowly getting there.

closed 2025-04-30 by Please-just-dont