wasm-micro-runtime/wasm-micro-runtime #4241: Allow malloc from shared heap without module_inst

calvin2021y, 2025-05-03:
#### Feature

Allow host to malloc from shared heap without module_inst, emplace the data into memory, then decide which module to detach.  

we also need a public api to tranlate the shared heap memory addr into module_inst int64 after attached.

#### Benefit

This is useful to avoid memory copy,  and the context is extract after the io task is done.   (context decidce which module_inst to call)

In this case,  after the io finished and data is copy into `shared heap`,  you has a chance to verify the data, then decide to create a new module_inst or use a exsits one to attach.

calvin2021y, 2025-05-04:
A useful feature  will be`bool wasm_runtime_attach_shared_heap(wasm_module_inst_t module_inst, wasm_shared_heap_t shared_heap, bool readonly);`

A readonly shared_heap make sure memory is readonly for `wasm_module_inst_t`

TianlongLiang, 2025-05-06:
> Allow host to malloc from shared heap without module_inst, emplace the data into memory, then decide which module to detach. 

You can use a preallocated shared heap to do that, it's not fully ready yet(In [this dev branch](https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/shared_heap) the interpreter support is finished and AOT is WIP)

You can check the example usage [here](https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/shared_heap/samples/shared-heap)

> we also need a public api to tranlate the shared heap memory addr into module_inst uint64_t addr after attached.

I believe that `wasm_runtime_addr_native_to_app` can do it.

> A readonly shared_heap make sure memory is readonly for wasm_module_inst_t

This can also be achieved by a preallocated shared heap, you can mmap a memory, then after finishing writing it use mprotect to make it read-only and create a preallocated shared heap from it.

yamt, 2025-05-08:
> > A readonly shared_heap make sure memory is readonly for wasm_module_inst_t
> 
> This can also be achieved by a preallocated shared heap, you can mmap a memory, then after finishing writing it use mprotect to make it read-only and create a preallocated shared heap from it.

assuming hw bound checks?

anyway, it sounds a bit fragile as wasm doesn't have a concept of read-only memory.