wasm-micro-runtime/wasm-micro-runtime #4263: read_wasm_binary_to_buffer() function is not available

kr-t, 2025-05-08:
Hello Everyone,

In https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/embed_wamr.md example, there are mention of following API:
```
/* read WASM file into a memory buffer */
  buffer = read_wasm_binary_to_buffer(…, &size);
```

Nevertheless, this function does not exists in the code, therefore making the documentation incomplete/misleading. I would recommend one of the following solutions:

1. Replace _read_wasm_binary_to_buffer()_ to _bh_read_file_to_buffer()_ in the documentation. POSIX example in product-mini uses it and it serves exactly the same purpose, as _read_wasm_binary_to_buffer()_ suppose to be serving. In this case I would also recommend including build steps for this file as in this issue: https://github.com/bytecodealliance/wasm-micro-runtime/issues/4221
2. Implement the _read_wasm_binary_to_buffer()_ function in the appropriate file. This is a simple function and even an implementation is  already available under the similar _bh_read_file_to_buffer()_ function.
3. Note in the documentation, that _read_wasm_binary_to_buffer()_ is not available and user shall write it by themselves. Examples of implementation can be included (few implementations can be found via Google).

Let me know what do you think is the appropriate fix for this. I would gladly help with the PR, if needed.

raiderss, 2025-05-09:
Brilliant implementation! At Eyes Store, we're also exploring similar technologies for our FiveM Tebex marketplace.

TianlongLiang, 2025-05-12:
I think the first approach would be a good choice, it would be great if you are willing to help

closed 2025-05-23 by kr-t