wasm-micro-runtime/wasm-micro-runtime #4272: extended-const

yamt, 2025-05-12:
it would be nice to implement [extended-const](https://github.com/WebAssembly/extended-const) as it's getting more important.

* [lime1](https://github.com/WebAssembly/tool-conventions/blob/main/Lime.md#lime1) has it enabled
* it benefits dynamic libraries (dynamic linking is not a thing for wamr yet though)

yamt, 2025-05-13:
an implementation is not trivial because the current logic assumes that we can transform const exprs to a constant value at load/compile time. (the assumption is not entirely correct even without extended-const.)
i guess a straightforward implementation involves aot format changes.

lum1n0us, 2025-05-15:
@kylo5aby Please help us manage it. ☕

closed 2025-07-07 by lum1n0us